opentrack/opentrack #1789: Nuralnet tracker missing from 2023.3.0 installer

zomgugoff, 2024-01-20:
The Opentrack installer does not include the Neuralnet plugin. It's only in the portable release.

sthalik, 2024-01-21:
See if this build works: https://ananke.misaki.pl/opentrack/opentrack-test-20240121_01.7z

This is a candidate for 2023.3.1, because more things are broken.

zomgugoff, 2024-01-21:
The portable version of 2023.3.0 was not an issue. It was the opentrack-2023.3.0-win32-setup.exe file that does not include the plugin.

stewi2, 2024-02-24:
Somehow the neuralnet tracker and a few others (kinect?) aren't showing up in the list of inputs, even using the above prerelease build. I do see the DLL (even using the released 2023.3.0 installer), but see no way to actually enable it: 
[image]

stewi2, 2024-02-24:
Looks like this has been going on for a while: https://github.com/opentrack/opentrack/issues/1394