opentrack/opentrack #1790: Joy binding doesn't work in Assetto Corsa

adi518, 2024-01-20:
Actually, I'm not sure if this is Opentrack to blame, but when I bind 'Center' to an unassigned gamepad button, it doesn't work in-game, unless I pop and focus the Opentrack window, which is... not fun. Ideas? I searched AC thoroughly to make sure the button I want to use is not bound to anything else that might override the binding to Opentrack. I also tried another unassigned button and it behaves the same.

sthalik, 2024-01-20:
I don't think we ever implemented XInput support, which is likely what your gamepad is using.

adi518, 2024-01-20:
Oh, wow thanks for the reply. I thought it was related to XInput somehow. Ideas, other than using Joy2Key?

closed 2024-01-20 by adi518