opentrack/opentrack #1792: Add Opentrack in Lutris wiki page (I have no perms)

raven-intheclouds, 2024-01-21:
Hello, I would like to add the following to a new "Opentrack in Lutris" wiki page (for wich I don't have perms)

# Opentrack in Lutris

One can successfully run Opentrack win32 portable version with Linux games through Lutris and Proton.

For that:

1. Download Opentrack win32 portable from the GH releases, and put it somewhere where Lutris can access it.
For example: `/home/deck/Games/IL2/gamefiles/opentrack/opentrack.exe`
2. Automatically start Opentrack before the game, and close it afterwards, by creating and selecting Lutris `pre-launch` and `post-exit` scripts.
3. Create a `home/deck/Games/pre-launch.sh` and give it executable perms:
```sh
#!/bin/sh

# Optional:
# create a reverse tunnel to the android device running HeadMob app (or similar) to reduce latency
# this needs gnirehtet in the path and working with the android device already.
# see: https://github.com/Genymobile/gnirehtet
#
# Deps: adb, gnirehtet, enable adb usb debugging in android phone
#
# Note: if using Lutris Flatpak, you need to use Flatseal to expose PATH to Lutris so it can
# find gnirehtet, adb.
# GNIREHTET_APK=/$HOME/.local/share/gnirehtet/gnirehtet.apk gnirehtet run -p 4242 &

# Start Opentrack for Windows on the same WINEPREFIX. This allows Opentrack to communicate
# with the game.
# Note: one must have configured Opentrack already to start when the game executable starts
# (if not, alt-tab and click start).
${WINE} **/home/deck/Games/opentrack/opentrack.exe**
```
4. Create a `home/deck/Games/post-exit.sh` and give it executable perms:
```sh
#!/bin/sh
pkill gnirehtet
pkill opentrack.exe
```
5. Open Lutris, game -> configure -> system options -> game execution -> set Pre-launch script & Post-launch script to the created files (don't check "wait for pre-launch script completion"). save.
6. Play. Voila!

## Examples

This is known to work at least for IL-2 1946, IL-2 Battle of Stalingrad (BoX).

theRealBassist, 2024-06-19:
Just want to add that this works perfectly for DCS as well!

raven-intheclouds, 2024-08-21:
Adding this info to the wiki has been unblocked by https://github.com/opentrack/opentrack/issues/1901.

raven-intheclouds, 2024-09-04:
It is also possible to use https://github.com/markx86/opentrack-launcher.
This python script crates a .bat executable that executes both the game and Opentrack via Proton steam, and it works with whatever arguments one is passing anyways. Here it works with several games.