opentrack/opentrack #1807: Missing many input options

SkippTekk, 2024-02-18:
OS: EndeavorOS Arch
Version: 2023.3.0-26-g3fe74769

Installed from Wiki

Not sure what's going on but trying to get this working for DCS and other applications.

[image]

SkippTekk, 2024-03-04:
Seems that the email method had failed. I'm trying to get the face tracking method working.

SkippTekk, 2024-03-04:
Ok, sorry the name of it is "neuralnet". that's the main one i would wish to get working.

berarma, 2024-03-04:
These are the dependencies directly from the code:

```
OpenCV
OpenMP
ONNXRuntime
```

If `tracker-neuralnet` isn't listed when running cmake then some dependency is missing.

SkippTekk, 2024-03-05:
Thank you! That's what i was missing :)

closed 2024-03-05 by SkippTekk

NickDeBeenSAE, 2024-07-28:
OS: Nobara40

Installed from the Github page instructions

This means that openmv does not exist.

I recommend removing this dependency.

``No match for argument: openmv``

sthalik, 2024-07-28:
`opencv` is missing, not openmp.

NickDeBeenSAE, 2024-07-28:
Open CV is fine.

It was not a weak.

sthalik, 2024-07-28:
`openmp` is needed so that `tracker/neuralnet` can set the amount of cores as per its settings.