opentrack/opentrack #1811: Kinect v2 support using libfreenect2

JustCryen, 2024-03-02:
Is this already implemented?
I've only found some information on the proprietary sdk, like in this issue https://github.com/opentrack/opentrack/issues/861 but nothing on the open one yet.
I'd like to use Kinect v2 for headtracking on Linux, is this currently possible?
If not, how hard would it be to implement a libfreenect2 based tracker for opentrack?

Slion, 2024-03-05:
It's not implemented. In theory you should be able to use the colour cam to run any colour based trackers. Though I don't know if that works out of the box on Linux. I guess you could modify the existing Easy tracker which works with the IR buffer of the Kinect to work on Linux. Sounds like just an integration job really. Should not be too hard, just time consuming.

JustCryen, 2024-03-05:
I just don't see my Kinect as an input device. I'm not sure if it's an issue on my side or if it doesn't show up as a webcam in general.