opentrack/opentrack #1817: Opentrack  problem!

kaiolavi, 2024-03-13:
The Opentrack window opens. But it doesn't connect the mobile device, and I don't understand where to look for the reason. Could someone advise a beginner.

kaiolavi, 2024-03-14:
I'm sorry if I'm so blind that I can't see the advice from the FAQ menu, I'd be grateful if someone could advise why opentrack doesn't connect a mobile device to a PC!

Pieloth, 2024-03-14:
Nobody seems to understand what mobile device you're trying to connect
Could you please give more information to be able to help you

kaiolavi, 2024-03-14:
The phone is a Samsung Galaxy A52, the SmoothTrack app was bought from the play store!

Pieloth, 2024-03-14:
Okay

Did you follow and apply all the settings instructions given on the SmoothTrack app page for PC and A52?

[https://smoothtrack.app/](https://smoothtrack.app/)

kaiolavi, 2024-03-14:
That's all I did
John Goering according to the instructions in the video

Pieloth, 2024-03-15:
Something probably wrong in the settings, but difficult to tell what

A possibility would be to use wireshark to dump traffic network between your A52 and PC, to check port 4444 UDP reception on PC

kaiolavi, 2024-03-18:
I got the Openrack program working, thanks for that.
Another question would be, why Opentrack does not open at all on a Win 11 machine.
I tried the latest version?