opentrack/opentrack #1818: Win 11 kill Opentrack!

kaiolavi, 2024-03-13:
I installed the last version of Opentrack on the Win 11 machine, and for a while it opened, but now it doesn't even open anymore, would using an older version help?

kaiolavi, 2024-03-16:
Even the Opentrack window does not open on this Win 11 machine, would the old version help with Win 11?

kaiolavi, 2024-03-18:
I got the Openrack program working, thanks for that.
Another question would be, why Opentrack does not open at all on a Win 11 machine.
I tried the latest version?

sthalik, 2024-03-18:
Try this build first: https://github.com/opentrack/opentrack/issues/1394#issuecomment-1938864870. If it still won't work, downgrade to 2023.1.0.

sthalik, 2024-03-18:
Answered in #1817.

closed 2024-03-18 by sthalik

kaiolavi, 2024-03-18:
Thank you ,i must try!