opentrack/opentrack #1819: Very high power consumption and CPU load when tracking is enabled.

SAVANNist, 2024-03-16:
[image]
[image]
as soon as I start tracking, the power consumption increases almost 10 times, the load increases from 10% to 70%, and the CPU temperature indicator shows 59 degrees (in games up to a MAXIMUM of 50). 10 cores, 3.5GHz. Help.

Vampir0, 2024-03-18:
I have the same problem, but it helped me to lower the CPU use to a quarter by dedicating only one thread to tracking, in the Neuralnet Tracker configuration, under Tuning/Debug

[image]

mjcarter, 2024-03-24:
Think this is normal if you're using Neuralnet, it is pretty CPU intensive and not any way to fully avoid that other than running Opentrack on a server PC and connecting to it via UDP. Point tracking uses way less CPU (and so also has much less input latency)