opentrack/opentrack #1821: Toggle While Held: Mouse Right Button

Sir-Ironbeard, 2024-03-17:
Greetings,

I was wondering if there was a way to bind Toggle While Held to the Right Mouse Button, as this would be my preferred control scheme. If not, would this be implemented in the future?

sthalik, 2024-03-17:
There is. Additionally bind "toggle" to a key and press it once.

Sir-Ironbeard, 2024-03-17:
Thank you for the quick response! I might not understand you correctly. When I press the Right Mouse Button during the binding, nothing registers.
[image]

tomiko23lol, 2025-01-21:
We want the same thing. I want to be able to bind any key / gamepad / moue button to these shortcuts. 
In this case ( ARMA3, ARMA reforger) I want to disable tracking so in my case I want to use Zero white holding and bind it to "RightMouseButton"  while aiming ans focusing trough weapon optics.

It is totally useless when there is filter what can and can not be used as bind key.

Please add this feature already, It is requested repeatedly over and over.