opentrack/opentrack #1822: Mac OS no longer supported?

Eymcdn, 2024-03-18:
I fly Xplane 12 on an M2 Mac mini pro. I saw a video on YouTube about this tracking software and went to explore it in detail. It's advertised as supporting Mac OS, but for some reason I can't find a download for Mac, and I believe I read that Mac is no longer supported. It would be wonderful to have this available again. Is anyone able to help?

moe-sc, 2024-03-22:
Seems to be unmaintained since 2016. Could still try the last Version i could find, doubt [2.3.0 prerelease 6](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3-rc49p6), tho i doubt it will work on M2.

matatata, 2024-04-18:
Well I have put in some effort and you can use opentrack on macOS to some extend:

https://github.com/matatata/opentrack/releases/tag/opentrack-f109134