opentrack/opentrack #1826: "fatal error: windows.h: No such file or directory" when compiling with Wine support

YosypC, 2024-03-28:
I used ccmake to switch the boolean "SDK_WINE" on, then make, just after configuring and generating inside ccmake:

```
Consolidate compiler generated dependencies of target opentrack-compat
[ 14%] Built target opentrack-compat
Consolidate compiler generated dependencies of target opentrack-options
[ 21%] Built target opentrack-options
Consolidate compiler generated dependencies of target opentrack-api
[ 22%] Built target opentrack-api
Consolidate compiler generated dependencies of target opentrack-video
[ 24%] Built target opentrack-video
[ 24%] Built target i18n-module-video
[ 24%] Built target i18n-module-api
[ 24%] Built target i18n-module-compat
Consolidate compiler generated dependencies of target opentrack-csv
[ 25%] Built target opentrack-csv
[ 25%] Built target i18n-module-csv
Consolidate compiler generated dependencies of target opentrack-cv
[ 28%] Built target opentrack-cv
[ 28%] Built target i18n-module-cv
Consolidate compiler generated dependencies of target opentrack-spline
[ 30%] Built target opentrack-spline
Consolidate compiler generated dependencies of target opentrack-filter-accela
[ 31%] Built target opentrack-filter-accela
[ 32%] Built target i18n-module-filter-accela
Consolidate compiler generated dependencies of target opentrack-filter-ewma
[ 34%] Built target opentrack-filter-ewma
[ 34%] Built target i18n-module-filter-ewma
Consolidate compiler generated dependencies of target opentrack-filter-hamilton
[ 35%] Built target opentrack-filter-hamilton
[ 35%] Built target i18n-module-filter-hamilton
Consolidate compiler generated dependencies of target opentrack-filter-kalman
[ 36%] Built target opentrack-filter-kalman
[ 36%] Built target i18n-module-filter-kalman
Consolidate compiler generated dependencies of target opentrack-filter-nm
[ 38%] Built target opentrack-filter-nm
[ 38%] Built target i18n-module-filter-nm
Consolidate compiler generated dependencies of target opentrack-qxt-mini
[ 41%] Built target opentrack-qxt-mini
Consolidate compiler generated dependencies of target opentrack-logic
[ 46%] Built target opentrack-logic
Consolidate compiler generated dependencies of target opentrack-migration
[ 49%] Built target opentrack-migration
Consolidate compiler generated dependencies of target opentrack-pose-widget
[ 51%] Built target opentrack-pose-widget
Consolidate compiler generated dependencies of target opentrack-user-interface
[ 58%] Built target opentrack-user-interface
[ 58%] Built target i18n-module-user-interface
[ 58%] Built target i18n-module-logic
[ 59%] Built target i18n-module-migration
Consolidate compiler generated dependencies of target opentrack-version
[ 59%] Built target opentrack-version
Consolidate compiler generated dependencies of target opentrack-executable
[ 62%] Built target opentrack-executable
[ 62%] Built target i18n-module-executable
[ 62%] Built target i18n-module-options
[ 62%] Built target i18n-module-pose-widget
Consolidate compiler generated dependencies of target opentrack-proto-fgfs
[ 63%] Built target opentrack-proto-fgfs
[ 63%] Built target i18n-module-proto-fgfs
Consolidate compiler generated dependencies of target opentrack-proto-libevdev
[ 65%] Built target opentrack-proto-libevdev
[ 65%] Built target i18n-module-proto-libevdev
Consolidate compiler generated dependencies of target opentrack-proto-udp
[ 67%] Built target opentrack-proto-udp
[ 68%] Built target i18n-module-proto-udp
[ 69%] Generating opentrack-wrapper-wine.exe.so
In file included from /home/yosyp/Documents/Git/opentrack/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:15:
/home/yosyp/Documents/Git/opentrack/opentrack/compat/shm.h:10:10: fatal error: windows.h: No such file or directory
   10 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
winegcc: Segmentation fault (core dumped)
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Error 139
make[1]: *** [CMakeFiles/Makefile2:2443: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

Running Mint 21.3

Coecoenut, 2024-03-30:
According to https://github.com/opentrack/opentrack/issues/1560 you need to install wine-stable-dev. Did you try that?
I'm running on Debian stable and have libwine-development 4.2 and libwine-dev 8.0 installed, not sure which is required for OT.

YosypC, 2024-03-30:
`wine-stable-dev` was the solution. It needs to be included in the documentation, somehow. Thank you kindly.

closed 2024-03-30 by YosypC