opentrack/opentrack #1827: Wine plugin 9.5 breaks Opentrack/TrackIR head tracking, works in 9.4

Fell, 2024-03-28:


Fell, 2024-04-15:
I figured out that it doesn't work because opentrack-wrapper-wine.exe.so segaults with wine-9.5 but not with wine-9.4. Unfortunately I don't know how to debug it.

Fell, 2024-04-15:
Actually, the reason `opentrack-wrapper-wine.exe.so` doesn't start is not a segfault, but something weirder.

Opentrack attempts to run a command that looks somewhat like this:
OTR_WINE_PROTO=3 WINEESYNC=1 WINEFSYNC=1 wine /usr/libexec/opentrack/opentrack-wrapper-wine.exe.so

Which, in Wine 9.4 results in:
(no output, program keeps running)

But in Wine 9.5:
Application could not be started, or no application associated with the specified file.
ShellExecuteEx failed: File not found.

If I create a symlink like opentrack-wrapper-wine.exe -> opentrack-wrapper-wine.exe.so then the program stays open again, but it doesn't seem to work for tracking.

If I rename other programs from .exe to .exe.so they seem to run fine. 

I tried git bisect, but it just told me this:
wine: '/home/fell/Games/dcs-world' is a 64-bit installation, it cannot be used with a 32-bit wineserver.

If I try to use a 32-bit prefix, it always prints the "Application could not be started" message, even with wine-9.4

I'm so confused. Does that make sense to anyone?

Fell, 2024-04-23:
I did some more testing. I can confirm it's also broken in wine 9.6 and wine 9.7. However, it works fine with Wine 9.6 in WOW64 mode! I strongly believe the problem is related to this bug: https://bugs.winehq.org/show_bug.cgi?id=56483

I can also confirm that compiling OpenTrack with `OPENTRACK_WINE_ARCH=-m64` does not fix the issue.

sthalik, 2024-04-25:
If the bug you linked is in fact the one causing this issue, then you can adjust the way Wine is invoked as a workaround in https://github.com/opentrack/opentrack/blob/master/proto-wine/ftnoir_protocol_wine.cpp#L125.

Fell, 2024-05-07:
This issue has been fixed in Wine 9.8

closed 2024-05-07 by Fell