opentrack/opentrack #1829: 2023.3.0 - Relative Translation- incorrect 'relative' axis?

Keebler750, 2024-03-31:
### Discussed in https://github.com/opentrack/opentrack/discussions/1805

<div type='discussions-op-text'>

<sup>Originally posted by **Keebler750** February 11, 2024</sup>
[[POST #2 may be a better description of general issue]]
I'm LOVING the stability of tracking now that I'm using this setup, but there doesn't appear to be any documentation for the rather cryptic RT parameters and I'm having a tough time troubleshooting.

I have to ask - what is 'relative translation' supposed to be relative to? 

When using the native Tobii apps, relative translation is relative to the current view angle so that when you look to the side and move forward, your ingame view moves toward the view vector.

When using OpenTrack, the translations appear relative to the nose of the aircraft no matter what I try to do. When I look to the side and move my head forward as above, you can see the camera shifting toward the front of the plane, not toward the view vector.

Is this a setup problem on my end, or is this a known bug?

INPUT: Tobii Eye Tracker
OUTPUT: freetrack 2.0 enhanced; Track IR disabled.
FILTER: Hamilton
(Tobii Game Hub exited in System Tray)</div>

annibale-x, 2024-06-02:
> ### Discussed in #1805
> Originally posted by **Keebler750** February 11, 2024 [[POST number 2 in [DISCUSSION][(url)](https://github.com/opentrack/opentrack/discussions/1805#discussioncomment-8965887) may be a better description of general issue]] I'm LOVING the stability of tracking now that I'm using this setup, but there doesn't appear to be any documentation for the rather cryptic RT parameters and I'm having a tough time troubleshooting.
> 
> I have to ask - what is 'relative translation' supposed to be relative to?
> 
> When using the native Tobii apps, relative translation is relative to the current view angle so that when you look to the side and move forward, your ingame view moves toward the view vector.
> 
> When using OpenTrack, the translations appear relative to the nose of the aircraft no matter what I try to do. When I look to the side and move my head forward as above, you can see the camera shifting toward the front of the plane, not toward the view vector.
> 
> Is this a setup problem on my end, or is this a known bug?
> 
> INPUT: Tobii Eye Tracker OUTPUT: freetrack 2.0 enhanced; Track IR disabled. FILTER: Hamilton (Tobii Game Hub exited in System Tray)

I have the same problem: When I turn my head to look behind my plane and lean to the side as I would naturally do in real life, I find myself looking at my seat, contrary to what I expect to happen. The same movement, however, with Tobii's Game Hub does what I expect: I turn to the left to look behind, for example, and move my body to the left to look over the headrest of my seat.

Keebler750, 2025-03-18:
I have now tested this with a Grass Monkey 240hz tracker and the same issue exists. It is not just Tobii related.

OpenTrack is not using the proper view vector but the longitudinal aircraft vector, which is not correct. Relative translation is messed up and this is getting zero attention for some reason.

Keebler750, 2025-03-24:
There seems to be an issue here, but it's more complex than I originally thought, and it's hard to nail down the variables without understanding expected behaviour. I'm closing this issue as I can't separate the camera issues from the OpenTrack issues.

closed 2025-03-24 by Keebler750