opentrack/opentrack #1835: build instructions for macOS

matatata, 2024-04-11:
Hi,

since I've worked a bit on some smaller issues on the macOS build I sometimes  get requests of users not being able to use my macOS build https://github.com/matatata/opentrack/releases, because they're not signed.

Perhaps you could add some instructions in the README, so they interested MAC users can build opentrack themselves: Here's I am able to successfully build a macOS opentrack:

Prerequisits: Install macports

export PATH=$PATH:/opt/local/bin:/opt/local/libexec/qt5/bin
export OTR_OSX_ARCH=arm64 # this is probably not necessary when building on Intel-Mac
sudo port install qt5 cmake ImageMagick
git clone --recurse-submodules https://github.com/opentrack/opentrack.git
cd opentrack cmake -S . -B build --toolchain cmake/apple.cmake
cd build
make install
open install # in this folder you should find the opentrack.app

Greetings, Mattata

matatata, 2024-04-18:
I'll first try meson around with the linux build instructions. I did not know about ccmake and I'm currently trying to get the modules that rely on  opencv to build

matatata, 2024-04-18:
OK that helped. The linux instructions helped a lot. Setting the OpenCV-Dir is crucial. Now I have a PointTracker! This is how it worked for me:

Install macports https://www.macports.org/

export PATH=$PATH:/opt/local/bin:/opt/local/libexec/qt5/bin
# export OTR_OSX_ARCH=arm64 # build arm64 . For intel set to x86_64 or remove
sudo port selfupdate
sudo port install qt5 cmake ImageMagick create-dmg opencv4
git clone --recurse-submodules https://github.com/opentrack/opentrack.git
cd opentrack

export OpenCV_DIR=/opt/local/libexec/opencv4/cmake
cmake -S . -B build --toolchain cmake/apple.cmake
cd build
make install

matatata, 2024-04-20:
I finally managed to make a build that even includes the neuralnet-tracker. See https://github.com/matatata/opentrack/blob/master/macosx/README.md

It is great to be able to be able to track using a regular webcam on macOS too!

batur, 2024-06-05:
Hey, are there any updates about installation, especially without the Macports?

matatata, 2024-06-08:
> Hey, are there any updates about installation, especially without the Macports?

If you just want to use opentrack on macOS without compiling it yourself you can simply download the unsigned binaries I did build: https://github.com/matatata/opentrack/releases

I needed macports only for building the app from source code.

Note that they don't come with the same amount of plugins/features as the windows version.

Veydn, 2024-06-28:
> Note that the macOS builds I did don't come with the same amount of plugins/features as the windows version

Hey, that's the issue I'm running into. Your macOS build https://github.com/matatata/opentrack/releases works (thank you for that), but it doesn't have the 'Freetrack 2.0 Enhanced' output option. 

I'm trying to use Opentrack for WarThunder on my M1 Macbook, but war thunder seems to require the Freetrack 2.0 output. 

Is there a way I can add this plugin? I tried to build Opentrack myself following the instructions above, but I couldn't get it to work.

sthalik, 2024-06-29:
Use the Wine output method on non-Windows to get the equivalent of the freetrack output method.

Veydn, 2024-06-30:
> Use the Wine output method on non-Windows to get the equivalent of the freetrack output method.

Thanks for the suggestion! In this build I only have the "FlightGear, UDP over network, Virtual joystick, and X-Plane" output options. Is there a way for me to add the Wine method?
[image]

sthalik, 2024-06-30:
You need to set the `SDK_WINE` boolean and have `winegcc` in the PATH when building it.

matatata, 2024-07-21:
on macOS we apparently do not have a convenient way install the wine-sdk  at least I did not find it. One can install wine using homebrew, but it seems it does not contain the development files. I guess we'd need to ask the maintainer of wine for macOS which is https://github.com/Gcenx I think.

sthalik, 2024-07-22:
You can just compile it regularly and install to `/opt/wine`.

Veydn, 2024-07-22:
I tried that but unfortunately wasn't able to compile & get it to work successfully (I also don't really know what I'm doing and only follow guides I find here or other places)

james-baker-ucm, 2024-08-04:
I've managed to compile wine and get the wine sdk and winegcc etc, and have changed the SDK boolean, but unfortunately I'm having build errors that seem to be to do with the compat module (see image).

Any idea what might cause these or how to resolve them?
[image]

matatata, 2024-10-01:
I have the same error after building wine when trying to build with SDK_WINE. I can overcome the compiler-errors by specifying -DUSE_WS_PREFIX in the wing++ command, but it will complain that it cannot find and link libraries.
ld: library 'advapi32' not found. And yes, there's no library with that name in /us/local/lib/wine/x86-64-unix. It's just not built on macOS. There's the dll in  /us/local/lib/wine/x86-64-windows however, but afaik, that wine-wrapper-thing  is a shared library with both posix and windows calls that's sneaked in by wine later.

sthalik, 2024-10-01:
@james-baker-ucm right before `#include <windows.h>` try adding:

```c++
#undef htons
#undef htonl
```

@matatata try replacing `-m32` with `-m64` here: https://github.com/opentrack/opentrack/blob/master/proto-wine/CMakeLists.txt#L20

matatata, 2024-10-02:
I installed wine-stable from macports but in the "dev" variant. `sudo port install wine-stable +dev`. Then all the libraries are there! I had to build it from source which took ages, but now I was at least able to compile the wrapper... the third-party dll need to be copied to opentrack.app/Content/MacOS/Plugins and then it does work!!!!

matatata, 2024-10-05:
> @james-baker-ucm right before `#include <windows.h>` try adding:
> 
> ```c++
> #undef htons
> #undef htonl
> ```
> 
> @matatata try replacing `-m32` with `-m64` here: https://github.com/opentrack/opentrack/blob/master/proto-wine/CMakeLists.txt#L20

Ah thank you I’ll try that instead of the other option. If it works I’ll update the pull request that I had already opened.

matatata, 2024-10-05:
The #undefs did not help unfortunately. Seems USE_WS_PREFIX is used to avoid conflicts between winsock.h and the unix headers. There's not much to be found on the internet, but I've seen others use this define with wineg++ too. I think for the wine-wrapper sockets are irrelevant so I'd stick with that. The built wrapper works perfectly on macOS anyway.

JLCvzs, 2025-05-27:
Hey @matatata what a great release you did here. I was looking exactly for something like this on Mac and you release worked great on my Intel Mac.

As this is something I'm using for a college project and I want to add OSC output I was trying to build it with your version of opentrack but in my build I'm not getting neuralnet on the input available besides I'm doing this:

Install macports https://www.macports.org/

export PATH=$PATH:/opt/local/bin:/opt/local/libexec/qt5/bin
export OTR_OSX_ARCH=x86_64
sudo port selfupdate
sudo port install qt5 cmake ImageMagick create-dmg opencv4
git clone --recurse-submodules https://github.com/matatata/opentrack.git
cd opentrack

export OpenCV_DIR=/opt/local/libexec/opencv4/cmake
cmake -S . -B build --toolchain cmake/apple.cmake
cd build
make install

I was wondering if you can help me with this

[image]

matatata, 2025-06-01:
Hey @JLCvzs ,

getting neuralnet to compile took me a while as well. Make sure you also install  `libomp`. But that might not be enough. Last time I built the software I had to also use a different compiler than Apple-clang. It was clang-19 by that time.
Take a look at the cmake file of the neuralnet module and you'll see what requirements it has. It will silently skip the build if the requirements are not met (could simply mean something can't be found or somethings picky.... ).

# switch to clang-19 (install it first)
sudo port -N install clang-19
sudo port select --set clang mp-clang-19

<do the build>

# switch back to default compiler:
sudo port select --set clang none

Hope it helps.