opentrack/opentrack #1839: Neuralnet - hits about 60fps, but the game data seems to be capped at 30

JaPalen, 2024-04-18:
Hi

Having a problem with neuralnet, i've recently purchased a 60fps capable webcam to get a smoother track via neuralnet. 
However, when inspecting the settings, I can clearly see that the tracker is running at around 60fps, and the camera preview also reflects that, however the preview above game data makes it seem like 30fps. Feels like that in games like DCS too. Are there any other settings i should try? 
Im using Accela as the filter by the way.

[image]

Kingmilton7, 2024-05-16:
how did you even get yours that high im stuck at 5 fps with 60 fps camera

DaWelter, 2024-05-16:
You should also decrease the resolution. See my response to @Kingmilton7 : https://github.com/opentrack/opentrack/discussions/1849

> how did you even get yours that high im stuck at 5 fps with 60 fps camera

More powerful CPU probably and using 4 threads.

As to answer the question: I don't really know. Indeed the tracker loops runs at nearly 60fps. Maybe @sthalik has an idea. Do you have the option to try one of the other camera-based trackers?