opentrack/opentrack #1840: Debian Linux pitch rotation stutters in DCS

Wanchos, 2024-04-27:
Hi i have a problem using my delanclip to track my head in DCS. I can bind my yaw and pitch but the yaw is set as JOY_RY and pitch as JOY_Z. The JOY_Z seems to be the only axis to stutter while the other ones work fine any ideas? I am using libevdev as output and i saw that debian uses an older version of the package than other distros like arch do you think this could be the issue?

Wanchos, 2024-04-27:
So obviously the axis is output wrongly and if i try with steam input then my pitch becomes the left and right trigger. I probably have to change the output? So when i press the configure button for libevdev it says that it needs read write permission for /dev/input/uinput. Does that mean i have to put opentrack in the root group and give it read write privileges? Thanks for any help!
[image]

Wanchos, 2024-04-28:
So i tried opentrack in Euro truck simulator 2 and the there was no problem so it might be a DCS problem

sthalik, 2024-05-09:
> it says that it needs read write permission for /dev/input/uinput. Does that mean i have to put opentrack in the root group and give it read write privileges? 

No. 

Your distro might already be taking care of it. First try 

```console
ls -l /dev/input/uinput
```

There might already be an `input` or similar group to add your user into. However, the device node might not exist at all and instead be located at `/dev/uinput` (in which case it needs a symlink).

In the short term,

```console
chmod 666 /dev/input/uinput
```

and see if it works at all ingame.

Then you can start figuring out how to add a udev rule which is the proper solution here.

Wanchos, 2024-05-22:
Hello thanks for your fast answer. For me uinput is in /dev/uinput. I tried to chmod but have the same issues do you have any recommended solution for me?

sthalik, 2024-05-23:
Try this as root:

```console
ln -s /dev/uinput /dev/input/uinput
chmod 666 /dev/input
```

Wanchos, 2024-05-23:
> Try this as root:
> 
> ```
> ln -s /dev/uinput /dev/input/uinput
> chmod 666 /dev/input
> ```
I tried your suggestion and dcs could not detect any of my devices (hotas, opentrack etc) after running the commands. Is there something else that i could try? Btw i also want to clarify that my opentrack install is in my home folder just in case that is relevant.

sthalik, 2024-05-23:
Was DCS previously able to use your HOTAS controllers? Is it the case that making `uinput` available suddenly makes it unable to detect anything?

Wanchos, 2024-05-23:
Yes DCS was previously able to use my hotas controllers and detect opentrack. Making uinput available in the way you suggested seems to make DCS unable to detect anything.

sthalik, 2024-05-23:
You could try making a virtual joystick using the `xboxdrv` program and see if DCS detects it, without doing the `uinput` thing that I suggested earlier.

Wanchos, 2024-05-23:
I will give xboxdrv a try when I have some time otherwise I will try to see if the same thing happens on other distros. In worst case I will have to install windows I guess. Thanks for your time and help!

Wanchos, 2024-05-25:
So i tried using xboxdrv and it now works great thanks so much for your help!!!

closed 2024-05-25 by Wanchos

sthalik, 2024-05-25:
Are you able to open opentrack's evdev joystick inside DCS after starting xboxdrv? Otherwise, how are you sending the data?

Wanchos, 2024-05-26:
After starting opentrack and then my script DCS automatically detects opentrack as a xbox controller and I can freely assign the axes whichever way I want to use them. Does that answer your question?

sthalik, 2024-05-26:
This is good to know. Thanks for coming back and explaining it. 👍