opentrack/opentrack #1842: DelanCam libusbK driver issues

ninailsgimp, 2024-05-06:
Hello.
I read that the open driver for head tracking (Delancam, trackIR, etc.) was a lot less buggy/choppy. So, I uninstalled CL eye test and ps3 eye driver. I then followed the instructions here: https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions
Now, in my device manager the Delancam is listed under libusbK usb devices, as Delancam (interface 0)
Opentrack (2023.1.0) is only showing OBS virtual cam, and PS3 eye universal. Not the Delancam. 
How do I get my Delancam working/visible with Opentrack now?
I would appreciate any help. Thanks in advance.

sthalik, 2024-05-08:
When you select the "PS3 Eye open driver" it should run on the physical camera you've got from Delan.

Delanclip, 2024-06-11:
DelanCam1 does not require any drivers as it uses inbuilt Windows drivers to work correctly.
The only correct instructions for using DelanCam1 with Delanclip are available on our website: https://delanclip.com/manuals/

sthalik, 2024-06-11:
So the Delancam is an UVC Video device and despite that, the user can't get it listed in the PT tracker? This seems more like an OS issue than something opentrack is responsible for.

Delanclip, 2024-06-11:
Hi Staszek,

It seems that @ninailsgimp might have missed our manual from the very beginning and tried to use non-dedicated information to run DelanCam1, which led to issues with our camera drivers using libusbK. 

There shouldn't be any problems if the libusbK driver wasn't installed initially. DelanCam1 is a plug-and-play device that all systems recognize as a generic camera without any issues. It works seamlessly on Linux, and we've even received reports of it functioning well on macOS. 😊

Best regards,
Tom @ Delanclip