opentrack/opentrack #1853: next release (2024.1.0)

sthalik, 2024-05-25:
Hey @DaWelter,

Do you think it's a good time to make a release with the neuralnet and the new filter?

@GO63-samara 

Maybe you'd like to voice an opinion as well.

DaWelter, 2024-05-25:
Hi @sthalik,

In general yes. Let me add the latest network model and play test the new stuff a bit.

GO63-samara, 2024-05-25:
Yes, of course it's time!

YosypC, 2024-05-26:
Will this release include the camera pitch offset? I remember years ago when I used to use this software, it was present. I have my webcam above my monitor and there's clearly a skewed relationship between my head and the output, and it's very frustrating.

sthalik, 2024-05-26:
Unlikely. Do you need it for rotation or position correction?

I've implemented it many years ago but have since removed it. Since then I've forgotten how to implement it. Is it meant to un-rotate before relative translation and mapping?

YosypC, 2024-05-26:
I... don't know. I have a slight downwards pitch and I need to correct both translation and rotation axis as it's misinterpreted.

DaWelter, 2024-06-02:
@sthalik , go ahead with the release if you want. 
Looks like my recent results play just as well as `head-pose-0.2-big.onnx`. (Bit better accuracy on the test set but apparently also more susceptible to eye movements again ...)

sthalik, 2024-06-04:
See if this build works: https://ananke.misaki.pl/opentrack/opentrack-test-20240604_01.7z

Especially if you've had trouble running the last release due to DLL initialization errors.

Lorkys, 2024-06-04:
Hello

DLL Error

```
c:\HeadTrack>
DEBUG [C:\repos\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library C:\\HeadTrack\\opentrack-test-20240604_01\\install\\modules\\\\opentrack-tracker-kinect-face.dll: DEBUG [C:\repos\opentrack\api/plugin-support.hpp:106]: library "neuralnet" failed: "Cannot load library C:\\HeadTrack\\opentrack-test-20240604_01\\install\\modules\\\\opentrack-tracker-neuralnet.dll:

```

Again, no neuralnet in input list
PC: win 10 x64,MVS 2022

sthalik, 2024-06-05:
@Lorkys could it be that you CPU doesn't support AVX?

Lorkys, 2024-06-06:
> @Lorkys could it be that you CPU doesn't support AVX?

Yes, Do not support. :(
[image]

But, build from [https://github.com/opentrack/opentrack/issues/1394#issuecomment-997421729](url) present and working
[image]

Is it possible to combine these two incompatible things?

sthalik, 2024-06-06:
@Lorkys In this new build you need to delete `onnxruntime.dll` and rename `onnxruntime-noavx.dll` to `onnxruntime.dll`.

https://ananke.misaki.pl/opentrack/opentrack-test-20240607_01.7z

sthalik, 2024-06-06:
Get the new build https://ananke.misaki.pl/opentrack/opentrack-test-20240607_02.7z.

Then delete `onnxruntime.dll` and rename `onnxruntime-noavx.dll` to the aforementioned `onnxruntime.dll`.

Lorkys, 2024-06-07:
> Get the new build https://ananke.misaki.pl/opentrack/opentrack-test-20240607_02.7z.
> 
> Then delete `onnxruntime.dll` and rename `onnxruntime-noavx.dll` to the aforementioned `onnxruntime.dll`.

This Build Worked!!! big THANKS

Moreover, I took this library and replaced it into the previous release. It also worked.
[image]

How can I get this library independently of you (so as not to bother you every time)? Download? Build from source?

sthalik, 2024-06-07:
It's going to be included in the new release.

sthalik, 2024-06-07:
@DaWelter can you see if there's a performance improvement with `tracker/nn` using the build from <https://github.com/opentrack/opentrack/issues/1853#issuecomment-2153575536>? It's for release notes.

Other than that, the release build has been prepared at https://ananke.misaki.pl/opentrack/opentrack-2024.1.0-win32-portable.7z and I'm going to release it within the next few days as long as there are no last-minute issues (please report as usual).

sthalik, 2024-06-07:
The release build has been prepared at https://ananke.misaki.pl/opentrack/opentrack-2024.1.0-win32-portable.7z and I'm going to release it within the next few days as long as there are no last-minute issues (please report as usual).

@DaWelter can you see if there's a performance improvement with `tracker/nn` now that onnxruntime's been updated to 1.18?

Lorkys, 2024-06-07:
Kinetic-face dll error. AVX support?

[image]

The neural network is present and working. big THX again.

Performnce neural on my Old_School_CPU average 10%;

sthalik, 2024-06-07:
You need the Kinect SDK installed for Kinect.

sthalik, 2024-06-09:
@DaWelter what to do?

sthalik, 2024-06-09:
@renzhefeilong can you check whether switching to the 0.2 network helps anything?

[image]
[image]

renzhefeilong, 2024-06-09:
> @renzhefeilong can you check whether switching to the 0.2 network helps anything? [...]

Yes, using a 0.2 network, head-pose-0.2-big and head-pose-0.2-small abnormalities, head-pose normal.

> @renzhefeilong can you check whether switching to the 0.2 network helps anything? [...]

> @renzhefeilong can you check whether switching to the 0.2 network helps anything? [...]

Yes, using a 0.2 network, head-pose-0.2-big and head-pose-0.2-small abnormalities, head-pose normal.

DaWelter, 2024-06-11:
Oh oh did I break it?! Looks like it. Let me check.

sthalik, 2024-06-15:
Thank you for the last-minute warning, @renzhefeilong. I've downgraded it to 1.16.3 and put out the release. Please contact us again in case of problems.

renzhefeilong, 2024-06-15:
> Thank you for the last-minute warning, @renzhefeilong. I've downgraded onnxruntime to 1.16.3 and put out the release. Please contact us again in case of problems.

You've worked hard! Downloaded and tested, but the problem persists.

Keebler750, 2024-06-15:
As I have stated elsewhere in Discussions and also Issues, the implementation of "Relative Translation" is either bugged or certain hardware is not fully supported. Either way, the documentation for that feature is pretty sparse, and it would be nice to understand what the intended result is supposed to be for each setting, and what the translation is supposed to be "Relative" to. As is, in my notes, the translation appears to be relative to the longitudinal aircraft axis instead of the view vector, and both the ISSUES and the DISCUSSION section of this repo gets very selective traffic. :shrug:

sthalik, 2024-06-21:
@renzhefeilong the problem should no longer exist with @DaWelter's update of the old models. See this build: https://ananke.misaki.pl/opentrack/opentrack-test-20240621_01.7z.

Additionally there should be an improvement in performance to `tracker/neuralnet` due to it being a 64-bit build (32-bit isn't properly supported by ONNXRuntime). However, it requires AVX and some of the legacy modules such as `proto/fsuipc` aren't available anymore.

renzhefeilong, 2024-06-21:
@sthalik Thanks for the update, the problem is solved and neuralnet works better than before with higher FPS and less CPU usage.

renzhefeilong, 2024-06-21:
> @renzhefeilong the problem should no longer exist with @DaWelter's update of the old models. See this build: https://ananke.misaki.pl/opentrack/opentrack-test-20240621_01.7z.
> 
> Additionally there should be an improvement in performance to `tracker/neuralnet` due to it being a 64-bit build (32-bit Intel isn't properly supported by ONNXRuntime). However, it requires AVX and some of the legacy modules such as `proto/fsuipc` aren't available anymore.

Replacing the old versions with the new head-pose-0.2-big.onnx and head-pose-0.2-small.onnx works, and renaming onnxruntime-noavx.dll to onnxruntime.dll supports old CPUs, but not as well as the new versions.

sthalik, 2024-06-22:
There's no non-AVX version for the yesterday's 64-bit build. In fact the entire opentrack now requires AVX.

Should I make a legacy build?

renzhefeilong, 2024-06-23:
> There's no non-AVX version for the yesterday's 64-bit build. In fact the entire opentrack now requires AVX.
> 
> Should I make a legacy build?

Many people have given feedback on this issue, and I think it is still necessary.

sthalik, 2024-06-25:
TODO

- fix pseye-subprocess.exe
- add missing tracker/hatire to 32-bit build
- rebuild qt with -force-debug-info

YosypC, 2024-07-06:
Don't forget #1860!

sthalik, 2024-07-27:
So there are several major issues. I'll be working on resolving them.

- - - - - 

I can't compile Qt kde/5.15 with the newest MSVC compiler 17.10.5 (even after disabling most optimization flags and LTO). Downgrading Qt kde/5.15 to the revision used by previous opentrack releases doesn't help.

```console
Generating Code...
        link /NOLOGO /DYNAMICBASE /NXCOMPAT /DEBUG /OPT:REF /INCREMENTAL:NO /DLL /SUBS
YSTEM:WINDOWS /VERSION:5.15 /MANIFEST:embed /OUT:C:\repos\qt\qtbase\build\msvc64\lib\Q
t5Widgets.dll @D:\Temp\nm99D6.tmp
   Creating library C:\repos\qt\qtbase\build\msvc64\lib\Qt5Widgets.lib and object C:\r
epos\qt\qtbase\build\msvc64\lib\Qt5Widgets.exp
qlabel.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObj
ect const QWindowsStyle::staticMetaObject" (?staticMetaObject@QWindowsStyle@@2UQMetaOb
ject@@B)
qlineedit.obj : error LNK2001: unresolved external symbol "public: static struct QMeta
Object const QWindowsStyle::staticMetaObject" (?staticMetaObject@QWindowsStyle@@2UQMet
aObject@@B)
qwidgettextcontrol.obj : error LNK2001 [...]
```
- switch to Qt 6?
- switch the compiler to Clang?
- figure out what is actually wrong with "undefined reference to QMetaObject" during the build of QWidgets.

Video input with video/opencv via DShow / opencv_videoinput module doesn't work for some users
- downgrade opencv?
- switch MSVC back to `-MT`?

PS3 Eye open driver doesn't work for some users.
- downgrade libusb?
- switch libusb and opentrack to `-MT`?

IslandJohn, 2025-02-22:
I would like to help test some of these changes. Are some of these interim builds still available somewhere?

izolini, 2025-04-18:
DEBUG [C:\repos\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library C:\\Games\\Tools\\OpenTrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."

Is it right to have 4 slashes at modules\\\\opentrack-tracker-kinect-face.dll ?