opentrack/opentrack #1854: Trouble getting Neuralnet tracker working on Linux Arch

JosefFinke, 2024-05-26:
Hi, I have been trying to get face tracking with Neuralnet working on Linux for a couple days now and most online forums and guides simply say: Install onnxRuntime and include it during cmake and you are good.
Though i have tried that a lot of times and it hasn't worked for me thus far. Does not show up as an option in the input list.
I'm building from this repo: "https://github.com/opentrack/opentrack"
And I am including OnnxRuntime in cmake:
[image]

And OnnxRunntime is installed: 
[image]

Someone able to guide me here ? I m sure i am just making a mistake somewhere, why else would everyone else say its easy.

Cheers !

immakyle, 2024-05-26:
Same issue here. After configuration and make install, running opentrack results in...

DEBUG [/home/immakyle/opentrack/api/plugin-support.hpp:106]: library "easy" failed: "Cannot load library /home/immakyle/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-easy.so: (libQt6Test.so.6: cannot open shared object file: No such file or directory)"
DEBUG [/home/immakyle/opentrack/api/plugin-support.hpp:106]: library "neuralnet" failed: "Cannot load library /home/immakyle/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-neuralnet.so: (libmpi.so.40: cannot open shared object file: No such file or directory)"

Assuming I need to point to these libraries, but I also only half know what I'm doing.

sthalik, 2024-05-26:
```console
$ pkgfile -u
$ pkgfile libmpi.so.40
```

sthalik, 2024-05-26:
Why would `tracker/easy` link against qt6-test?

DaWelter, 2024-05-26:
Hm .. @JosefFinke  does the directory structure match `<onnx dir>/lib/onnxruntime.so`?
Here is what I have:
```
//Root directory of the ONNX Runtime installation
ONNXRuntime_DIR:PATH=/opt/onnxruntime-linux-x64-1.6.0

//Path to a file.
ONNXRuntime_INCLUDE_DIR:PATH=/opt/onnxruntime-linux-x64-1.6.0/include

//Path to a library.
ONNXRuntime_LIBRARY:FILEPATH=/opt/onnxruntime-linux-x64-1.6.0/lib/libonnxruntime.so
```

You could remove the `QUIET` option in  `tracker-neuralnet/CMakeLists.txt` to get error messages.

It's possible that our find script `cmake/FindONNXRuntime.cmake` doesn't work for you.

JosefFinke, 2024-05-26:
@DaWelter .. in the folder the screenshot is showing has only some cmake files in it. Though what I also tried later was grabbing the .tgz from their git releases and extracted the packages to the same folder opentrack lives in and pointed the cmake config to that. Which includes the /lib/onnxruntime.so file.

This is what I grabbed from them, maybe wrong version ?
[image]

Did also remove the QUIET otion from the cmakelists.txt, no error messages.

CannonFodderSE, 2024-05-29:
Did you ever get it to compile and work?
I'm having the same issue on the latest version of Kubuntu.
What is the correct source to download onnxruntime?

immakyle, 2024-05-29:
> Did you ever get it to compile and work? I'm having the same issue on the latest version of Kubuntu. What is the correct source to download onnxruntime?

Compile, yes. Work, no...
Onnxruntime was packaged in arch repo [here](https://archlinux.org/packages/extra/x86_64/onnxruntime/). You can follow the upstream url. 

Issue I'm having now is locating/pointing to Steam Proton's wine binary. I've been sifting through directories to no avail. Found a temporary solution by using this script [here](https://github.com/markx86/opentrack-launcher). It runs windows portable version of opentrack through proton. Works great. 

However, I'm still in pursuit of getting the linux version working...

DaWelter, 2024-05-29:
> Though what I also tried later was grabbing the .tgz from their git releases and extracted the packages to the same folder opentrack lives in and pointed the cmake config to that. Which includes the /lib/onnxruntime.so file

That should work. I assumed `opentrack-tracker-neuralnet.so` is not built. Can you find it in `<install folder>/libexec/opentrack/`?

Gontran08, 2024-10-02:
I have the same issue on Manjaro. The file is there yet I get an error on startup:

[image]

[image]

DaWelter, 2024-10-03:
Your `onnxruntime` seems to be linked against some MPI library. It looks like it due to the error from opentrack. Run `ldd` to check.

Assuming that's the issue ... MPI is a standard/interface for distributed computing. It wouldn't be needed for OpenTrack. The `pacman` output lists `openmpi` which is a specific implementation of MPI as *optional* dependency. So there could be the connection.
However, If mpi was really required, the package manager should have installed it automatically. Otherwise I'd expect it to work.

Gontran08, 2024-10-03:
I installed openmpi and now I can see neuralnet tracker in Opentrack. It seems to have fixed the issue.

theNizo, 2024-11-23:
What ultimately helped me was the comment from user Jiube [here](https://aur.archlinux.org/packages/opentrack).

Instead of `cmake ..`, run

```
cmake -DONNXRuntime_DIR=/usr/lib/cmake/onnxruntime -DONNXRuntime_INCLUDE_DIR=/usr/include/onnxruntime ..
```