opentrack/opentrack #1860: Reinstate a camera offset option

YosypC, 2024-05-31:
Many version ago OpenTrack had the ability to implement an offset in pitch to account for different camera positions. Many have their cameras up on the monitor, often pitched down. Without additional corrections, the output will be skewed and will result in movements in axis which do no exist, for example: when pitched down, head Z movement will result in negative Y translation.

I've opened a new issue to continue the #1853 discussion.

DaWelter, 2024-06-02:
I'm in favor of it. 

I guess it's on our/my todo list. Last year I made a draft for rotation correction only, included in this PR https://github.com/opentrack/opentrack/pull/1643 and didn't follow up on it. Maybe it's about time to revisit it. 

Ofc pull requests are welcome ;-)

Cat-Lady, 2024-06-13:
Yes, please. The unwanted pitch during yaw is quite annoying, and it seems to be 100% related to camera position (above/under) and camera fixed pitch.

Nels0, 2024-07-21:
Hi, I'm interested in this functionality too, or more specifically an arbitrary transform so that what's observed from the camera is remapped be relative to the screen's center.  Hoping to contribute code myself, but I'm still getting up to speed and getting build working.

A diagram because I'm not familiar with jargon, and excuse axis directions being inaccurate.
i.e: head moving in arrow direction will result in a translation with only a "screen (orange axes)" Z-component, whereas from the camera it would result in a translation with a Z and Y component.
and: monkey turning its head to the left resulting in rotation around the "screen" y-axis, whereas from the camera it would rotate around Y and X(?)

[image]

IslandJohn, 2025-02-22:
This is the same #639, no? One or the other can probably be closed.

YosypC, 2025-03-08:
> This is the same #639, no? One or the other can probably be closed.

It looks like so.

sthalik, 2025-07-31:
You guys won. I can no longer resist. It's on the shortlist f or 2025.1.

closed 2026-01-14 by YosypC

DaWelter, 2026-01-15:
What @Pieloth said, and make sure to have no center pose set while you do that. Disable center on startup.

You can try moving your head forward and backward and tweak the offset angles so that mostly the z value is changing.

The offset angles are important. The translation is just added to your pose, so it's mostly "cosmetic".