opentrack/opentrack #1866: Memory overrun when using pointtracker

Raven3-14, 2024-06-08:
While using opentrack with arma 3, it crashes after a few minutes (3~10?, haven't timed it).
I started it through commandline to monitor debug stuff, got a OutOfMemoryError when it crashed

output from commandline:
```
PS C:\Users\%myuser%> E:\opentrack-2023.3.0-win32-portable\install\opentrack.exe
PS C:\Users\%myuser%>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library E:\\opentrack-2023.3.0-win32-portable\\install\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
[ INFO:0@7.700] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
[ INFO:0@7.730] global videoio_registry.cpp:244 cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(4, sorted by priority): MSMF(1000); DSHOW(990); CV_IMAGES(980); CV_MJPEG(970)
[ WARN:0@33.191] global cap.cpp:344 cv::VideoCapture::open VIDEOIO(DSHOW): backend is generally available but can't be used to capture by index
[ WARN:0@72.125] global cap.cpp:344 cv::VideoCapture::open VIDEOIO(DSHOW): backend is generally available but can't be used to capture by index
DEBUG [F:\dev\opentrack\csv\csv.cpp:97]: csv: lookup game id 7503
DEBUG [F:\dev\opentrack\csv\csv.cpp:97]: csv: lookup game id 7503
[F:\dev\opentrack-depends\opencv\modules\core\src\alloc.cpp:73] opencv: Failed to allocate 2079604 bytes at cv::OutOfMemoryError
```

I tested changing process priority and core affinity earlier, had no effect on crashes

I am using windows 11, 16GB of ram, mostly full when running arma and other stuff
TrackIR protocol in freetrack for output, pointracker for input
Input camera source is a UVC gadget webcam, 1920x1080@45fps, only shows 20~30fps in tracker information

I believe the error could be caused by frames being generated faster than they are being processed, or by old frames not being discarded properly

sthalik, 2024-06-08:
This is super crazy, because the PT tracker hasn't had any leaks in over a decade.

> cap.cpp:344 cv::VideoCapture::open VIDEOIO(DSHOW): backend is generally available but can't be used to capture by index

First off, use this newest build that's meant to become the next release version. There was a change to camera backend selection. https://ananke.misaki.pl/opentrack/opentrack-2024.1.0-win32-portable.7z. Then the above quoted error shouldn't occur anymore.

opentrack is a 32-bit executable and presently it can use up to 2 gigabytes of memory.

What camera are you using? In case you're using the PS3 Eye, are you using the open driver or the CL-Eye driver?

See whether the memory consumption (use working set, not private working set) grows steadily as the program runs, or maybe it jumps by a large amount at once. When tracking a 640x480 image, I normally have the entire process use 90-100 MB working set and that value almost doesn't change during the entire lifetime of tracking. This will provide information on the exact class of this bug.

Once opentrack has crashed, don't close the error window but open the Task Manager and create a minidump.

[image]

If you compress the minidump into a `.zip` file, you'll be able to attach it to this issue by dragging it into the comment box (and thus not have to use any third-party service to share it).

Raven3-14, 2024-06-08:
I didn't get any error logs when it crashed, just a suspended opentrack.exe in task manager details that I couldn't kill, but seems to have self-closed overnight or after putting my computer to sleep

My camera is a bit odd, it displays as a UVC gadget to the computer and has only YUV420 output from what I can tell
It uses a Raspberry pi zero W 2 for processing and usb interface, I had frame rate issues or field of view issues with it running at anything but 1920x1080, but that shouldn't affect opentrack other than the higher resolution or the format.

ram used appears to be increasing continuously while running simple test (setting tracker to pick up maximum number of points, doesn't produce useful headtracking, but does seem to make it work fairly hard)
From details in taskmanager, only very rough memory values and times
10:24 ~400,000K
10:26 ~530,000K
10:28 ~600,000K
10:30 ~750,000K
10:32 ~900,000K

10:38 ~1,250,000K
10:44 ~1,600,000K
crashed between 10:44 and 10:48
Looks like a memory leak
Could it be that the higher resolution frames are being generated faster than they can be processed by pointtracker?

I'll try the newer version when I have more time (probably later today)

Can't upload the first memory dump from while running to github, even compressed it's 98MB, max allowed in comments is 25MB
Second memory dump, from after crash, was 1921 MB uncompressed

Resulting process in task manager, memory dump from it was ~2gb
[image]
Very similar output from commandline
```
PS C:\Users\%myuser%> E:\opentrack-2023.3.0-win32-portable\install\opentrack.exe
PS C:\Users\%myuser%>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library E:\\opentrack-2023.3.0-win32-portable\\install\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
[ INFO:0@10.802] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
[ INFO:0@10.838] global videoio_registry.cpp:244 cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(4, sorted by priority): MSMF(1000); DSHOW(990); CV_IMAGES(980); CV_MJPEG(970)
DEBUG [F:\dev\opentrack\tracker-pt\module\point_extractor.cpp:200]: camera driver doesn't support grayscale
[F:\dev\opentrack-depends\opencv\modules\core\src\alloc.cpp:73] opencv: Failed to allocate 2079604 bytes at cv::OutOfMemoryError
```

Raven3-14, 2024-06-09:
version 2024.1.0 still has a memory leak, 
Memory was growing slower with camera set to 15hz, and almost insignificantly with camera set to 5hz

Got a dump file from the 2024.1.0 release candidate version, had to be in volumes for uploading to github,
Size before zipping was 416,297 KB
[opentrack.001.zip](https://github.com/user-attachments/files/15754143/opentrack.001.zip)
[opentrack.003.zip](https://github.com/user-attachments/files/15754140/opentrack.003.zip)
[opentrack.002.zip](https://github.com/user-attachments/files/15754141/opentrack.002.zip)

This is the memory dump from 2023.3.0, after it crashed, 
Double zipped to get it to volumes that can be uploaded, size before zipping was ~2GB
[opentrackMemDumpAfterCrash_2.002.zip](https://github.com/user-attachments/files/15754154/opentrackMemDumpAfterCrash_2.002.zip)
[opentrackMemDumpAfterCrash_2.003.zip](https://github.com/user-attachments/files/15754156/opentrackMemDumpAfterCrash_2.003.zip)
[opentrackMemDumpAfterCrash_2.001.zip](https://github.com/user-attachments/files/15754157/opentrackMemDumpAfterCrash_2.001.zip)

sthalik, 2024-06-09:
Can you provide a *crash* dump for 2024.1.0? Thanks.

Raven3-14, 2024-06-10:
Screen turned off for a short while when getting it to crash, memory dropped to normal-ish amount, 190,000K in working set
(earlier memory values were just the active private working set, shared working set didn't change much though, looked like it stayed constant at ~43,000K)

~20min for crash from started to OutOfMemoryError
```
DEBUG [C:\repos\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library E:\\opentrack-2024.1.0-win32-portable\\install\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
[ INFO:0@9.649] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
[ INFO:0@9.672] global videoio_registry.cpp:244 cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(3, sorted by priority): DSHOW(1000); CV_IMAGES(990); CV_MJPEG(980)
[F:\dev\opentrack-depends\opencv\modules\core\src\alloc.cpp:73] opencv: Failed to allocate 2079604 bytes at cv::OutOfMemoryError
```
Memory dump from after crash of 2024.1.0
Unzipped size ~2GB, 
[opentrack2024_1_0MemDumpAfterCrash.003.zip](https://github.com/user-attachments/files/15770869/opentrack2024_1_0MemDumpAfterCrash.003.zip)
[opentrack2024_1_0MemDumpAfterCrash.002.zip](https://github.com/user-attachments/files/15770964/opentrack2024_1_0MemDumpAfterCrash.002.zip)
[opentrack2024_1_0MemDumpAfterCrash.001.zip](https://github.com/user-attachments/files/15770656/opentrack2024_1_0MemDumpAfterCrash.001.zip)

Raven3-14, 2024-06-23:
Memory leak is also present in neuralnet tracker in 2023.3.0, reached 900,000KB working set in ~8min,
1,200,000 ~13min after start
Possibly caused by something with opencv, or caused by a higher resolution camera than expected
I'll test with a different camera and with 2024.1.0 in the next couple of days

Tested the test tracker, no memory leaks occurred, unresolved allocations seemed to be part of opentrack.exe windows stuff

Managed to build opentrack from the master branch,
Version I built showed same memory leak, memory persisted when starting and stopping the tracker, didn't clear until opentrack was closed
Couldn't figure out the source of the memory leak, probably related to point tracker
Memory snapshot in visual studio showed the largest amount of memory was in unresolved allocations
The unresolved allocations seem to be stuff that isn't tracked by visual studio, they varied in size between 0B and ~530,000B

Raven3-14, 2024-06-23:
Did some more testing, issues seem to be partially caused by the UVC raspberry pi camera I'm trying to use. I tried with lower resolution down to 320x180
1280x720 was slower, odd camera issues, 
640x480 and 320x180 both ran near to the camera's frame rate (45hz)
Memory leak persisted with all output resolutions, 

Tested with the webcam built into my laptop, didn't see any memory leak issues, 
Tested the uvc camera using windows camera and vlc, memory leak only occurred when using opentrack with pointracker or neuralnet (only tested those two for camera input trackers though, I think they both use opencv)

Seems like it might be related to the camera I'm trying to use 
Do you have any suggestions for troubleshooting the camera/opentrack interactions?

sthalik, 2024-06-24:
I don't think any good memory leak debugging tools exist for Windows.

Raven3-14, 2024-06-24:
Did some further testing on 2024.1.0 with the uvc camera, 
Both easy tracker and arcuo tracker appeared to have memory leaks as well

After all these tests, it seems like the get frame function or similar used in all of them doesn't play nice with my camera
I'm probably just going to find a different camera or reprogram my raspberry pi webcam so I can use headtracking

The memory leak still exists under certain conditions (using a uvc-gadget powered raspberry pi webcam, that doesn't seem to support mjpeg output,) but all I've been able to figure out is it's probably related to the use of opencv, or the way the original camera get frame stuff was written.
I don't know if it would even be reproducible with any other camera.

DaWelter, 2024-06-24:
AFAIK all the camera based trackers use opencv's `VideoCapture` class.

sthalik, 2024-06-24:
See: https://github.com/opentrack/opentrack/blob/master/video-ps3eye/module.cpp#L216

sthalik, 2024-06-24:
See: https://github.com/opentrack/opentrack/blob/master/video-opencv/impl-camera.cpp#L62

Raven3-14, 2024-08-02:
Been a while since I looked at it,
I gave up with trying to rewrite the opencv capture library to fix the memory leak
Every combination of tracker and camera I tested seemed to show that it was a result of my camera

I successfully managed to use the raspberry pi camera with opentrack by using the virtual camera from OBS to separate the camera from opentrack
I've noticed no issues with this combination other than slightly higher CPU usage, and some advantage with being able to preprocess the video stream before opentrack gets it
It might have introduced a bit more latency, but not that I can notice with the camera and OBS running at 120fps, and the tracker running at 80+ hz most of the time