opentrack/opentrack #1868: opentrack crashes every time it is started

nocosimfarmer, 2024-06-13:
Having an issues using opentrack with a tobii 5 and Farming Simulator. Every time I start opentrack the app crashes.

nocosimfarmer, 2024-06-13:
Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x6548ea46
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x00000000
Faulting process id: 0x0xE904
Faulting application start time: 0x0x1DABDDADCBBBD6C
Faulting application path: C:\Program Files (x86)\opentrack\opentrack.exe
Faulting module path: unknown
Report Id: b317d8c9-a9b7-4463-abe4-e75b1cbd82a1
Faulting package full name: 
Faulting package-relative application ID:

nocosimfarmer, 2024-06-13:
Any help would be greatly appreciated!!

blackmagic1493, 2026-09-14:
Did you ever find out why?