opentrack/opentrack #1869: Star Citizen running in proton not receiving input

OmegaRogue, 2024-06-17:
I'm running Star Citizen from lutris with Proton GE 9.7, the tracking is showing the correct values in Opentrack, but the game seems to not be receiving the data from Opentrack. Wine prefix and executable are set to the same as used to launch the game in lutris.

OmegaRogue, 2024-06-17:
This seems to be caused by umu-run utilized by lutris when using proton

SkippTekk, 2024-06-22:
Did you get this resolved? I'm recently getting the same thing BUT also "waiting for game" in the top menu. Was curious if you fixed it as it might fix my issue on DCS

OmegaRogue, 2024-06-28:
> Did you get this resolved? I'm recently getting the same thing BUT also "waiting for game" in the top menu. Was curious if you fixed it as it might fix my issue on DCS

I haven't yet

DoomSlinger, 2024-07-16:
I found a temporary fix, (won't work with GE proton, or any proton that is installed in compatibilitytools.d).

You need to create a simlink in your compatdata folder to your star citizen (or any game for that mater) prefix folder and name it a random number that isn't used by another game (if a game has no dlc, you can use that games ID and replace the 0 with any number between 1-9)

For example, if your star citizen prefix is ~/Games/star-citizen (default when installing with lutris) Then you need to create a simlink to that folder in your compatdata.

Name the symlink a random number (in my case I used 15 but I don't know if that is used by any other steam game or program)

then in opentrack, select the version of proton you are using, and set the steam application id to the number you named the file as.