opentrack/opentrack #1877: 2024.1.1 PT/open driver issue?

Len62, 2024-06-24:
@sthalik 
Trying out the new version and both the 32 bit and 64 bit versions through up these error messages when tracking is started. Everything is still fine in previous versions.
[image]

I may try uninstalling the PS3 open driver and reinstalling it, to see if that helps.

sthalik, 2024-06-24:
See if replacing this file fixes the issue. I've downgraded libusb to the release version 1.0.27.

[ps3eye-subprocess.zip](https://github.com/user-attachments/files/15954692/ps3eye-subprocess.zip)

Len62, 2024-06-24:
Did not work for either version. I do notice that when I start tracking the camera turns on. I may go ahead and reinstall the PS3 open driver.

sthalik, 2024-06-24:
In case you've installed the driver using the old faulty procedure (reinstalling over and over again until it works), you've ended up with a different libusb kernel-side driver version compared to the new reliable procedure that most people are by now using. So reinstalling the Windows kernel driver might actually help.

Len62, 2024-06-24:
Ok, completely removed camera/driver. Reinstalled and tried both of these drivers:

[image]

Both worked fine with 2023 versions, but both throw up the same errors as before, even after replacing the ps3eye-subprocess file. Same for both the 32 and 64 bit versions.

Len62, 2024-06-25:
Thank you. I'll run through those today.

Len62, 2024-06-27:
Sorry I'm late getting back to you. I tried all of the versions from the zip file and none of them worked. I tried moving the version from the 2023.2.0 release, file size 281 KB, dated 5/27/23, and it worked! 2024.1.1 runs fine using that file. I did notice that if I ran those files directly I would get a pop-up window saying the device had no driver. I tried capturing the message but the pop-up closes to fast.

sthalik, 2024-06-27:
>  I did notice that if I ran those files directly I would get a pop-up window saying the device had no driver.

Can you get me the exact error message?

Len62, 2024-06-28:
By running ps3eye-subprocess_v1.0.23.exe from a cmd prompt I was able to get the error message:

libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_046D&PID_C092&LAMPARRAY\6&BB2008B&0&207834834733_SLOT00'
libusb: info [winusb_get_device_list] libusb will not be able to access it
libusb: warning [set_composite_interface] failure to read interface number for HID\HID_DEVICE_SYSTEM_VHF\7&2A22484&0&0000, using default value

petsild, 2024-06-29:
Hi,
I managed to install libusb and Opentrack works correctly.
When I look in the device manager, I see that the driver for the microphone is missing?

[image]

sthalik, 2024-06-29:
@ptsild Try installing libusbK in the libusb installer, then the microphone should work.

petsild, 2024-06-29:
Which driver is better for OT libusbk or libusb0?

sthalik, 2024-06-29:
libusbK should be marginally faster as long as it doesn't have compatibility issues on your system. Have to test it out by using it.

petsild, 2024-06-30:
I put libusbk in there and the device manager error went away, thanks.

petsild, 2024-06-30:
Is this correct for checking?

[image]

sthalik, 2024-06-30:
No. Interface 1 should have the built-in generic Windows driver available. You should have it listed when you select "show only compatible drivers" after selecting "update driver".

petsild, 2024-06-30:
I manually selected from the audio device list, thanks for your support.

[image]

sthalik, 2024-06-30:
Cieszę się, że sobie poradziłeś.
Jsem rád, že jsi to vyřešil.

sthalik, 2024-06-30:
So for the next release, the 2023.3 driver subprocess will be copied verbatim, but later opentrack versions must figure out why it isn't working for multiple users.

timmeys, 2024-07-13:
Today tried the 32 bit version 24.1.1 and everything works!!! as written put the driver camera sony...