opentrack/opentrack #1880: Saving Camera Settings

IrishDeviant, 2024-06-26:
When I go to the Options menu > Tracker Tab > Camera Settings and click Open, I set my settings, hit apply and ok. And they do apply to the camera in the video viewport. However, as soon as I hit OK on the Options dialog box, it takes a second or two to save, then the camera reverts back to default settings. So things like brightness, contrast, etc, all revert to default, making it useless at tracking. The only way I can get it to work now is to launch Opentrack, open options, set all of my camera settings, and leave the options window open while I play. Which is not ideal. It just started doing this too with no apparent changes before hand. It used to save the camera settings. 

Any thoughts on what could be causing this?

Hollywood83, 2024-06-26:
Yeah, same

sthalik, 2024-06-27:
Was it working in 2023.3.0, and is it newly broken in 2024.1.1?

Hollywood83, 2024-06-27:
> Was it working in 2023.3.0, and is it newly broken in 2024.1.1?

For mine I'm not sure when it quit working. I haven't had a lot of time to game lately. I didn't manually update before it quit working though unless the software auto updates. I did uninstall today and reinstall with whatever the latest version is while trying to fix it, but it didn't do anything.

sthalik, 2024-06-27:
Try downgrading opentrack until you find a version that saves camera settings properly.

Hollywood83, 2024-06-27:
> Try downgrading opentrack until you find a version that saves camera settings properly.

I'll do that this evening and let you know what I find out. Are there any files or folders I need to manually delete after uninstalling to fully clear it?

Hollywood83, 2024-06-28:
> Try downgrading opentrack until you find a version that saves camera settings properly.

So, this is strange. When I last closed opentrack, the camera settings had reverted to default. When I just launched it, the setting were my originally saved settings. I adjusted settings, hit apply, closed options, and the setting stuck. So, intermittent issue? It seems odd that something like saving a file could have an intermittent issue. I'm still in the current version, I didn't try downgrading yet. It was just working when I opened it.

sthalik, 2024-06-30:
There are actually two separate modes for using the camera settings button. One is when tracking is stopped and another one when it's running. Can you confirm that both modes are broken in the same way?

Hollywood83, 2024-07-03:
> There are actually two separate modes for using the camera settings button. One is when tracking is stopped and another one when it's running. Can you confirm that both modes are broken in the same way?

Hey sorry I didn't see your response. I just checked it and it appears to be saving with the camera started as well as stopped. I created a new profile so maybe it was a corrupt profile that was causing the issue.

Hollywood83, 2024-07-03:
> There are actually two separate modes for using the camera settings button. One is when tracking is stopped and another one when it's running. Can you confirm that both modes are broken in the same way?

Is there a way to package a build with preset settings? I sell an ir cam that uses Opentrack, but have no programming ability. I'm more of a designer/hardware guy than software.

sthalik, 2024-07-04:
> Is there a way to package a build with preset settings? 

Yes. Stick a `default.ini` into `modules/presets`. It's going to get included in every new profile. Otherwise stick an `.ini` not named `default.ini` and it's going to get copied to user's local profile directory but allow him to delete it (according to file modification time).

sthalik, 2024-07-08:
@IrishDeviant can you answer [my question](https://github.com/opentrack/opentrack/issues/1880#issuecomment-2198681444)? Need to figure out what's wrong with it in order to fix it. Thanks.