opentrack/opentrack #1885: opentrack crashes when I press options

lucasordenes, 2024-07-01:
The program opens fine, but as soon as I press "Options" or "Start" the program crashes immediately and It never closes. I tried everything, so I formatted my system, changed my webcam for a PS3cam and It worked fine until I pressed stop or something like that and It stopped working like before I formatted my drive... It's worth noting that I changed versions, deleted residual files, changed my camera again and used my phone's camera, went to a restore point from when I just formatted my pc, etc. Like I did everything that I could think of. The most frustrating thing is, is that It worked for like ten minutes after a lot of work and then I was back to square one. Anyone has any suggestions or ran into a similar problem?

lucasordenes, 2024-07-01:
[image]
It would stay like this until I reboot my system

lucasordenes, 2024-07-09:
So I figured it out what was the problem, It was very weird... I had to unplug my controller and now the programs works perfectly and in order to use opentrack with my controller, first I have to start tracking with opentrack and then plug my controller and everything works fine. Hope that helps someone.

closed 2024-07-09 by lucasordenes

MagicianBlackH, 2024-10-17:
Thanks, I have met the same situation. But I think it is troublesome to unplug and then plug my controller every time I want to use the opentrack.