opentrack/opentrack #1887: [Ubuntu 22.04] Neuralnet tracker successfully built, but does not appear in Input

leftoutofdark, 2024-07-05:
After having issues with building Neuralnet tracker, i was able to successfully get an ONNX directory registered and the Neuralnet tracker was successfully built, including a opentrack-tracker-neuralnet.so file in /libexec/opentrack. I have also installed qt6-base and openmpi as provided [here](https://github.com/opentrack/opentrack/issues/1854#issuecomment-2132058790l), however this has not resolved the issue. I apologise if I have simply made an error or this issue has already been resolved, but I have no clue where to go from here.

DaWelter, 2024-07-06:
Idk. @sthalik ? Just two ideas: 
* Start OpenTrack from the console and see if there is an error.
* Run `ldd` on the tracker `.so` to see if all the dependencies are found.

leftoutofdark, 2024-07-07:
Hm, after re-installing opentrack the issue has disappeared and I can now use the neuralnet tracker. However, I did install wine-stable-dev while re-installing opentrack, so perhaps the issue was that I didn't have that?

closed 2024-07-07 by leftoutofdark