opentrack/opentrack #1893: problem with libevdev joystick reciever

OracleToes, 2024-07-15:
This issue seems related #1120 
I've compiled on Arch from a cloned git of this project, and it seems to work mostly fine.
I'm trying to use this for head tracking for X4 Foundations, but it seems like X4 isn't picking up the tracking so i'm trying to use the joystick receiver. That sort of works, except the axes are all kind of random, some are joystick axes, some are triggers, and what ever the "down" output is, isn't getting picked up by X4 at all.
So I try to edit the mappings from opentrack, and I get the error "Make sure rw for /dev/input/uinput!"
On my system there is no /dev/input/uinput, but there is a /dev/uinput, so I symlinked it, and added RW permissions on both /dev/uinput and the symlink. On issue #1120 someone mentioned running `sudo modprobe uinput` which i have tried, but doesn't fix anything, I get the same error.
Am I doing something wrong here? I just want to be able to remap the axes of the output to something sensible.

walseb, 2025-05-06:
In case anyone stumbles across this, or if you are still facing this issue, the message `Make sure rw for /dev/input/uinput!` is actually just a hint, not an error message. It shows up like that even if the permissions & symlinks & kernel modules are properly set up.

To see whether OpenTrack works, simply run a utility like `jstest-gtk` and check wheter it detects the new input.

This message should probably be modified to make this clear. It fooled me causing me to debug for an hour.

skauertz, 2025-09-14:
OK, so while playing around a bit with the source code and looking into libevdev, I found that simply adding a (fake) vendor and product ID to the joystick device definition in `/proto-libevdev/ftnoir_protocol_libevdev.cpp` makes it get recognized like a proper joystick on Linux and not as a gamepad, which resolves the problem of having no "down" input in certain games. 

As a bonus, Elite:Dangerous no longer "forgets" the mapping to the devices axes since it does not find a VID/PID.

Simply add this after libevdev_set_name and recompile:

```
    libevdev_set_name(dev, "opentrack headpose");
    libevdev_set_id_vendor(dev, 1);
    libevdev_set_id_product(dev, 1);
```

sthalik, 2025-09-18:
@skauertz can you confirm that the merged change on branch `fix-libevdev-unrecognized`works for you? I've committed it under your name. Is there something that still needs changing?

skauertz, 2025-09-18:
Hello @sthalik - thanks for the quick reply.

I confirm that the patch on that branch works well. 

To me that should be sufficient to fix this issue. There are perhaps more appropriate VID/PID to use, e.g. I found [VID 1209 "Generic"](https://devicehunt.com/search/type/usb/vendor/1209/device/any), but no "Generic" PID to go with it. It would probably work with any VID/PID you assign, as long as it's not empty.