opentrack/opentrack #1896: unable to choose Proton-Steamplay-version in Wine-Windowslayer for Unix - dropdown empty

TechGameGeek, 2024-07-20:
Hi,
just compiled opentrack for linux with "wine - windowslayer for unix". It is working great so far e.g. XPlane 12 (linux native), but I am unalbe to pass the trackingdata to steam/proton - games.

As I think to know, I have to choose wine/proton in "wine-windowslayer for unix" and set "Proton Steamplay" to the corresponding version, the game runs with (via dropdown) and give the steam-app-id of the game in the box "steam-app-id"

Unfortunatly my dropdown (Proton Steamplay) is empty.

So I assume I have missed some packages during compiling?

Can you please tell me, what I did wrong to not see "Proton Steamplay" versions?

Thanks!

sthalik, 2024-07-25:
Probably Valve again broke things.

You can add your path to the list here: https://github.com/opentrack/opentrack/blob/master/proto-wine/proton.cpp#L17-L27.

spikerguy, 2024-08-10:
> Probably Valve again broke things.

Not really, the path is correct. I can see only few proton versions only.

[image]

Though I have Proton 8 and 9 also.

sthalik, 2024-08-10:
Where is the Wine prefix located?

spikerguy, 2024-08-10:
> Not really, the path is correct. I can see only few proton versions only. [...]

> Where is the Wine prefix located?
```
/$USER/Games/SteamLibrary/steamapps/compatdata/805550/pfx/
```

This is game specific, as per the gui wine prefix is supposed to be when wine is used 

for steam proton it just needs proton version and app id, is this information correct ??

I think I will be documenting the whole gaming on linux situation and that way can contribute some documentation for proton use on open track

sthalik, 2024-08-10:
Try adding it here: https://github.com/opentrack/opentrack/blob/master/proto-wine/proton.cpp#L18-L20

What you need added is probably `/Games/SteamLibrary/steamapps/compatdata/805550/pfx/` as the pathname is preceded by the user's home directory. If that doesn't work then there's some trial and error involved.

This should become fixed once #1879 is implemented (in indeterminate future).

spikerguy, 2024-08-10:
I see a good documentation [here](https://forum.endeavouros.com/t/linux-gaming-guide/7339)

> Proton engines location:
> 
> ~/.steam/steam/steamapps/common
> 
> Keep in mind that Proton will be installed in the same library as the first Steam Play game you try to launch, and will only be installed in one location.
> 
> Game PREFIX location, where [appid] is your game id (look up [steamdb 25](https://steamdb.info/) if in doubt):
> 
> ~/.steam/steam/steamapps/compatdata/[appid]/pfx/

sthalik, 2024-08-10:
This is straightforward to implement. 

- find all `steamapps/compatdata` locations, for each of them:
- - find all `appid` subdirectories, for each of them:
- - - find and register all prefix subdirectories

spikerguy, 2024-08-10:
> This is straightforward to implement.
> 
>     * find all `steamapps/compatdata` locations, for each of them:
> 
>     * * find all `appid` subdirectories, for each of them:
> 
>     * * * find and register all prefix subdirectories

Oh I found Proton 8.0 in different directory

`/home/user/Games/SteamLibrary/steamapps/common/`

so it have proton in compatedata as well as common. this could be the issue. I think a shared memory should resolve it

Thanks

spikerguy, 2024-08-10:
I applied these patches on opentrack-git from aur and still it is not able to dedect all proton versions

```
--- a/opentrack/proto-wine/proton.cpp	2024-08-10 17:58:40.462649519 +0300
+++ b/opentrack/proto-wine/proton.cpp	2024-08-10 18:03:45.508109393 +0300
@@ -18,6 +18,8 @@
     "/.steam/steam/steamapps/compatdata",
     "/.local/share/Steam/steamapps/compatdata",
     "/.steam/debian-installation/steamapps/compatdata",
+    "/Games/SteamLibrary/steamapps/common",
+    "/Games/SteamLibrary/steamapps/compatdata",
 };
 
 static const char* runtime_paths[] = {
```

also tried 

```
--- a/opentrack/proto-wine/proton.cpp	2024-08-10 17:58:40.462649519 +0300
+++ b/opentrack/proto-wine/proton.cpp	2024-08-10 18:03:45.508109393 +0300
@@ -18,6 +18,8 @@
     "/.steam/steam/steamapps/compatdata",
     "/.local/share/Steam/steamapps/compatdata",
     "/.steam/debian-installation/steamapps/compatdata",
+    "/home/gamer/Games/SteamLibrary/steamapps/common",
+    "home/gamer/Games/SteamLibrary/steamapps/compatdata",
 };
 
 static const char* runtime_paths[] = {
```

does it need proton version and app id both ? or can it just run in appid's pfx ?

sthalik, 2024-08-10:
As a workaround you can add `/Games/SteamLibrary/steamapps/compatdata/805550`

TechGameGeek, 2024-08-11:
@sthalik thanks for your reply, I will try it as soon as possible

CaptPickguard, 2024-09-06:
Running flatpak Steam and having the same issue. Did try compiling manually and adding the paths, but the dropdown never did populate.

Resorted to using the Windows version of Opentrack within the Proton prefix for now.

Priton-CE, 2025-01-19:
@CaptPickguard for flatpak we had issue #1941 

flatpak is just hard to work with and at least in the past also had other issues.