opentrack/opentrack #1897: cmake installation causes a sandbox violation in Gentoo

kakra, 2024-07-21:
The Gentoo build system is sandboxed and doesn't allow writes to the final system destination during install. Instead, all writes must go to the temporary image directory which Gentoo will then copy to the live system (checking for conflicting files etc).

Opentrack's build system violates this sandbox:
```
 * ----------------------- SANDBOX ACCESS VIOLATION SUMMARY -----------------------
 * LOG FILE: "/var/tmp/portage/app-misc/opentrack-9999/temp/sandbox.log"
 *
VERSION 1.0
FORMAT: F - Function called
FORMAT: S - Access Status
FORMAT: P - Path as passed to function
FORMAT: A - Absolute Path (not canonical)
FORMAT: R - Canonical Path
FORMAT: C - Command Line

F: chmod
S: deny
P: /usr/./share/opentrack/i18n
A: /usr/share/opentrack/i18n
R: /usr/share/opentrack/i18n
C: /usr/bin/cmake -P cmake_install.cmake

F: unlink
S: deny
P: /usr/./share/opentrack/i18n/nl_NL.qm
A: /usr/share/opentrack/i18n/nl_NL.qm
R: /usr/share/opentrack/i18n/nl_NL.qm
C: /usr/bin/cmake -P cmake_install.cmake
 * --------------------------------------------------------------------------------
```

This boils down to the following situation: The translation merger creates absolute system paths for the installer. Also, translation merging happens during the install phase rather than in the build phase. This is probably why you need to remove the `*.qm` files prior to regenerating: they are not properly tracked by the build system. Both, removal and installation cause a sandbox violation.

The problem can be seen in `cmake-install.cmake` which is used by Gentoo to install the project:
```
if(CMAKE_INSTALL_COMPONENT STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  list(APPEND CMAKE_ABSOLUTE_DESTINATION_FILES
   "/usr/./share/opentrack/i18n/nl_NL.qm")
  if(CMAKE_WARN_ON_ABSOLUTE_INSTALL_DESTINATION)
    message(WARNING "ABSOLUTE path INSTALL DESTINATION : ${CMAKE_ABSOLUTE_DESTINATION_FILES}")
  endif()
  if(CMAKE_ERROR_ON_ABSOLUTE_INSTALL_DESTINATION)
    message(FATAL_ERROR "ABSOLUTE path INSTALL DESTINATION forbidden (by caller): ${CMAKE_ABSOLUTE_DESTINATION_FILES}")
  endif()
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/./share/opentrack/i18n" TYPE FILE MESSAGE_LAZY PERMISSIONS WORLD_READ OWNER_WRITE OWNER_READ GROUP_READ FILES "/var/tmp/portage/app-misc/opentrack-9999/work/opentrack-9999_build/nl_NL.qm")
endif()
```

This message is only present for the qm files.

I'm currently unsure how to proceed from here. It's not as easy as patching some paths before or after the configure phase. I probably try to disable qm processing until this problem is resolved.

sthalik, 2024-07-23:
Try this patch (with `-p0`).

```diff
diff --git cmake/opentrack-i18n.cmake cmake/opentrack-i18n.cmake
index 6c424017..1f0c67d9 100644
--- cmake/opentrack-i18n.cmake
+++ cmake/opentrack-i18n.cmake
@@ -58,7 +58,8 @@ endfunction()
 
 function(otr_merge_translations)
     otr_escape_string(i18n-pfx "${opentrack-i18n}")
-    install(CODE "file(REMOVE_RECURSE \"\${CMAKE_INSTALL_PREFIX}/${i18n-pfx}\")")
+    #install(CODE "message(FATAL_ERROR \"foo \${CMAKE_INSTALL_PREFIX}\")")
+    #install(CODE "file(REMOVE_RECURSE \"\${CMAKE_INSTALL_PREFIX}/${i18n-pfx}\")")
 
     foreach(i ${opentrack_all-translations})
         get_property(ts-files GLOBAL PROPERTY "opentrack-ts-files-${i}")
@@ -90,7 +91,7 @@ function(otr_merge_translations)
         add_dependencies(i18n-lrelease ${target-name})
 
         install(FILES "${qm-output}"
-                DESTINATION "${CMAKE_INSTALL_PREFIX}/${opentrack-i18n}"
+                DESTINATION "${opentrack-i18n}"
                 PERMISSIONS ${opentrack-perms-file})
     endforeach()
 endfunction()
```

sthalik, 2024-07-23:
> Also, translation merging happens during the install phase rather than in the build phase. This is probably why you need to remove the *.qm files prior to regenerating: they are not properly tracked by the build system.

It's a workaround for dependency issues. It needs to work with 4 different CMake generators.

kakra, 2024-07-23:
> Try this patch (with `-p0`).

Yep that works but I had to convert it to a `-p1` patch. Thanks, this is great. :-)

So now I could continue and create German translations.

> It's a workaround for dependency issues. It needs to work with 4 different CMake generators.

You mean like ninja vs gmake vs the Windows stuff probably? Couldn't meson and ninja work here? cmake seems to be its very own beast.

sthalik, 2024-07-23:
Fixed, thanks. Please report other things you may run into.

closed 2024-07-23 by sthalik