opentrack/opentrack #1898: migrating from $HOME/.local prefix to system /usr prefix installation: wine-injection fails

kakra, 2024-07-22:
After testing opentrack in my local home prefix, I created a system package to install to `/usr`. With this installation, wine injection stopped working. If I run the locally compiled version (no matter if it's in my project folder or a tmp folder), it works again.

From the proton logs I can see that injection starts `FreeTrackClient64.exe` inside the prefix if it works. With the system-installed version it does not work: There's no error, the process simply won't get created. But I don't have any clue why. I can only imagine there's a problem with the wine wrapper but even opentrack doesn't show any error, it just states that the process has been started. It probably started and immediately quits again without injecting into wine.

Do you know where this can fail? The only difference is that my local prefix installation uses vanilla build settings while the system package builder uses a QA-verified build environment (it strips debug symbols by splitting them to separate files, applies some extra cflags, ensures some rpath stuff etc). I made it that my local prefix uses the same cflags but this doesn't change behavior.

I also tried removing the FreeTrack registry keys because they point to the wrong location after changing the install location, and they would be regenerated. But this really makes no difference: My local prefix installation doesn't care about changing the install location, it just works.

Things I want to try next:

1. replace the files installed by the system manually with those from the local prefix
2. strip installed files from debug symbols

I'm not sure if (2) makes a difference which leaves me with (1). But if that works, it only proves that something else went wrong in the system package builder.

So my questions is: Is there any pitfall with this you may have come across during development that I should inspect especially?

sthalik, 2024-07-22:
Your Wine may be using some kind of sandboxing. There's a registry key that points games toward `NPClient64.dll`. 

```
HKEY_CURRENT_USER\SOFTWARE\NaturalPoint\NATURALPOINT\NPClient Location -> Path
```

Here's an example value from my Windows machine: `F:/dev/opentrack/build/install/modules/`. 

You probably want to add a new hard drive via `winecfg` pointing to the root of the filesystem, or use `winepath(1)` and `strace(1)` to debug it.

kakra, 2024-07-22:
Yep, sandboxing indeed. Now that you mention it, it hits me like a ton of bricks: Proton executes within a bwrap container, it can access stuff in your home directory but only very selected locations of the system (like video drivers).

Testing:

1. Copying my system-built version to my home directory: everything works.
2. Copying my custom-built version to my system: nothing works.

No both builds are actually completely fine, it's the location that matters.

Now let's see how we can fix that. I'll let you know. I think opentrack can improve two things here: Add a log message if it cannot properly execute any of the components. And improve documentation, maybe even give a hint in the proto-wine settings dialog what needs to be done.

I'll come back here if I found a solution.

sthalik, 2024-07-22:
The solution is #1879.

kakra, 2024-07-22:
I don't think so because the Proton prefix won't contain any component of FreeTrack until you injected it, or does it?

The problem is starting `opentracker-wrapper-wine.exe.so` in the first place. What is "the DLL" you are talking about? Is it what the registry refers to, e.g. `FreeTrackClient64.dll`?

Then even that is the problem here: Proton (wine) inside the bwrap container cannot access this DLL if it is installed to `/usr`. Proton does see a system there but it's not my host system, it's a special runtime environment by Valve that mimics a special distribution tailored for Proton.

But I think it is possible to add configuration for Proton to allow access to additional files outside of this container: They already do that dynamically for graphics drivers.

kakra, 2024-07-22:
@sthalik Here's the explanation:
https://github.com/ValveSoftware/steam-runtime/issues/288#issuecomment-717954844

When injecting into Proton, it should check if `/usr/libexec/opentrack` actually exists, and if it doesn't, it needs to use `/run/host/usr/libexec/opentrack` instead.

kakra, 2024-07-22:
So the solution is probably pretty simple: I can get away with installing opentrack to `/opt/opentrack` because that is outside `/usr` in the Steam Runtime container, and thus available as-is. It's actually where I did my test installs initially, and it worked. I'll just create a symlink from `/usr/bin/opentrack` to `/opt/opentrack/bin/opentrack` and it will just work.

kakra, 2024-07-22:
It works that way:
```
# pgrep -a opentrack
683657 /usr/bin/opentrack
685429 Z:\opt\opentrack\libexec\opentrack\opentrack-wrapper-wine.exe.so
```

If you're curious, here's what the build script looks like:
https://github.com/gentoo/gentoo/pull/37646