opentrack/opentrack #1901: contributing to the wiki

sthalik, 2024-07-25:
You can now update the wiki by sending pull requests or direct commits to the https://github.com/opentrack/wiki repository.

The `opentrack/opentrack` [wiki](https://github.com/opentrack/opentrack/wiki) synchronizes 3 times a day.

Please try to avoid editing the `opentrack/opentrack` wiki directly (for project committers). The script tries to perform simple merges but conflicts can cause problems.

The script actually runs on my VPS rather than the GitHub infrastructure, but I'm going to provide it for transparency purposes after I'm sure that it doesn't have any subtle bugs.

raven-intheclouds, 2024-08-21:
Hi, thanks!

Please, consider just allowing on editing the Wiki via Web UI; it lowers the contribution barrier, and sometimes one doesn't have git configured (like me, in a gaming machine right now).

raven-intheclouds, 2024-08-21:
It is also possible to use https://github.com/markx86/opentrack-launcher.
This python script crates a .bat executable that executes both the game and Opentrack via Proton steam, and it works with whatever arguments one is passing anyways. Here it works with several games.