opentrack/opentrack #1902: Not working with the latest DroidCam Client

Credelle, 2024-07-25:
Opentrack is not working with the newer droidcam app, it is not detecting/showing the camera
[image]

sthalik, 2024-07-25:
Is this happening on both 32-bit and 64-bit versions?

Credelle, 2024-07-26:
> Is this happening on both 32-bit and 64-bit versions? Is this happening in 2023.3.0?

Happened on 2023.3.0, then i updated to 2024.1.1x64 and i still had the same issue

ojigoon, 2024-08-15:
I seem to be having the same issue... just downloaded DroidCam and Opentrack to try headtracking out, doesn't work with 2024.1.1x64.

xFluffyDemon, 2024-08-22:
Same issue here,but classic DroidCam pc client feeds video, but its limited to 30fps

joosniz, 2024-10-28:
I found a workaround for anyone using an Nvidia GPU. I ran into the same issue but I tried routing the camera feed through Nvidia Broadcast without applying any effects and that seems to make Opentrack able to pick it up. Unfortunately it's limited to 30 fps but it does work.