opentrack/opentrack #1908: I cannot seem to find NeuralNet at all in my input options

KPgreg07, 2024-08-03:
(Disclaimer - In no way am I a programmer, just trying to enhance my flight sim experience)

Hello,
I have recently installed opentrack on my laptop in order to use it when I use fsx. I have been able to get it to work with AI track, but the movement is very choppy. However, I found out about NeuralNet from different videos and was immediately interested, as I think that it might improve performance with one less application running. Unfortunately, I can't find it at all in the catalogue of inputs. Does anyone know anything on how to fix this?

Separate question, if someone knows how to reduce the choppiness of the head movement with both opentrack and AI track, could you let me know? Thanks!

kakra, 2024-08-06:
> Separate question, if someone knows how to reduce the choppiness of the head movement with both opentrack and AI track, could you let me know? Thanks!

You also need "ONNX Runtime" for NeuralNet to be available. But I'm not sure where to put the DLLs if you're running Windows.

To reduce choppiness in opentrack, try the following steps in this order:

- reduce camera resolution to 640x480
- increase light exposure
- disable camera dynamic framerate exposure (so the camera sticks to a fixed fps)
- ensure it's running at 30 fps or higher
- use the AccelaHamilton filter
- increase threads in NeuralNet, try up to 4 threads, inference time should be below 2-3ms
- try to adjust the smoothing and deadzone settings (carefully!) in the filter

Overall it's important that your CPU isn't too busy with the game already, otherwise opentrack has to fight over CPU resources with the game and this may look choppy. You can try to adjust opentrack's priority using task manager.

KPgreg07, 2024-08-06:
Okay, I installed the version that initially had neuralnet released and sure enough, it was there! IDK, it's there though!!

kakra, 2024-08-08:
Today, I found out that the Windows setup version ships with ONNX included (and thus neuralnet is available) but it still didn't work because some antivirus program on my friend's PC blocked access to the camera.

DaWelter, 2024-08-12:
> Okay, I installed the version that initially had neuralnet released and sure enough, it was there! IDK, it's there though!!

To be clear, what is there, only the DLLs or is the tracker loaded now?

Anyway, on windows `onnxruntime.dll` and `modules/opentrack-tracker-neuralnet.dll` should be installed in the opentrack dir.

sthalik, 2024-08-12:
He might be hitting that `-MT` issue from an older build. @KPgreg07 did you try the `2024.1.1` build?

KPgreg07, 2024-08-17:
@sthalik I will try again and see if this works!