opentrack/opentrack #1913: Feature request: Auto update function

d-kleine, 2024-08-08:
Would it be possible to add an auto update functionality into the app, so that app directly in itself without needing to looking for a new release on gh?

sthalik, 2024-08-10:
It's a good feature. Actually the TrackHat opentrack version already has it.

But opentrack as of right now doesn't phone home in any way. No telemetry, no background downloads, no nothing. Here an argument can be made that an update check is more valuable, but I'm not sure. With a GitHub account you can subscribe to a project's releases and receive them through email and that may be a suitable workaround at least for people already with GitHub accounts.

d-kleine, 2024-08-10:
Yeah, agree - would be a nice "quality of life" optional feature though. I mean, it could be disabled by default, and if you don't want it make it phone home for updates, just don't enable it 😅