opentrack/opentrack #1916: Opentrack not working with DroidCam

ojigoon, 2024-08-15:
I downloaded DroidCam and Opentrack to try out headtracking. Opentrack remains blank even though DroidCam displays video. Followed the quick start guide, and various tutorials to no avail.

[image]

vido89, 2026-10-13:
Same here Im on Linux Mint 22.3 and opentrack does not detect my camera while OBS studio/droidcam app does and scrcpy also does work

 scrcpy --video-source=camera
/snap/scrcpy/502/bin/scrcpy-launch: Warning: The "raw-usb" snapd security confinement interface is not connected, the OTG mode will not work.
/snap/scrcpy/502/bin/scrcpy-launch: Warning: The "joystick" snapd security confinement interface is not connected, the Gamepad mode will not work.
scrcpy 5.0.1 <https://github.com/Genymobile/scrcpy>
Camera video source: microphone audio source selected
INFO: ADB device found:
INFO:     -->   (usb)  XXXXXXXXXXX                     device  SM_A266B
/usr/local/share/scrcpy/scrcpy-server: 1 file pushed, 0 skipped. 502.5 MB/s (733930 bytes in 0.001s)
[server] INFO: Device: [samsung] samsung SM-A266B (Android 15)
[server] INFO: Using camera '0'
INFO: Renderer: opengl
INFO: OpenGL version: 4.6.0 NVIDIA 470.256.02
INFO: Trilinear filtering enabled
INFO: Trilinear filtering disabled for VA-API frames
INFO: Interop: vaapi
ERROR: [FFmpeg] Failed to initialise VAAPI connection: -1 (unknown libva error).
ERROR: Could not create VA-API device: Input/output error
INFO: Hardware decoding unavailable; using software decoding
INFO: Interop: software
INFO: Video decoding: software
INFO: Texture: 4080x3060

vido89, 2026-10-13:
Got it working 

```
wget https://github.com/Genymobile/scrcpy/releases/download/v5.0.1/scrcpy-linux-x86_64-v5.0.1.tar.gz
tar xf scrcpy-linux-x86_64-v5.0.1.tar.gz
cd scrcpy-linux-x86_64-v5.0.1
```

`./scrcpy --video-source=camera --camera-size=1280x720 --camera-fps=30 --v4l2-sink=/dev/video10 --no-playback --no-audio`

"$HOME/opentrack-install/bin/opentrack"