opentrack/opentrack #1917: Vjoy 2.1.9.1 virtual joystick not found when Joytick Emulation -- vJoystick selected

leslieb256, 2024-08-18:
When I run open track I can't output to vjoy because it does not seem to detect the vjoy joystick

I have installed vjoy 2.1.9.1.
I have installed OpenTrack 2023.3.0
I am using windows 11 Pro
Vjoy is working with joystick gremlin (so I assume it is ok).
From this [Issue #1700](https://github.com/opentrack/opentrack/issues/1700)  it looks like win11/2.1.9.1 is a combination that should work.

When I try to get to the options for vjoy I get this window:
[image]

I have tried resetting my vjoy config and uninstalling and reinstalling and also tried the Brunner Eletronik version of 2.2.2.0.
I also tried using version 2023.1.0 to which is the same version that was used in the post. And I tried running it as admin.
I also tried 2024.1.1 (both as normal user and admin)

I am happy to poke around to try to work out what is going on but I'd need a bit of guideance.

kiren665, 2024-08-20:
I have the same problem. I have tried the above method, but it didn't work for me. vjoy is running normally, but it doesn't seem to receive any signal from opentrack.

NathanielLLally, 2024-10-02:
Same issue.  Any reason joystick gremlin works fine but not opentrack?

Keltskiy, 2024-10-10:
Issue#1938 is not a dupe, different issue.
(Opentrack recognizes and successfully hooks vJoy#1, however opentrack fails to influence vjoy#1 axis.)

TheNextVision, 2025-09-19:
Was this ever solved?

DarkSynopsis, 2025-11-22:
Was just messing with this myself and vJoy vanished when I copied the dll but I had originally copied the vJoyInterface.dll from the x64 folder since that is what my shortcut lead to when I used it to just jump to the directory but you need the x86 one, working for me now.

Kuiriel, 2026-10-30:
> 7\. Was just messing with this myself and vJoy vanished when I copied the dll but I had originally copied the vJoyInterface.dll from the x64 folder since that is what my shortcut lead to when I used it to just jump to the directory but you need the x86 one, working for me now.

I am having the same issue as everyone else. Where did you get the x86 version of this dll from? 2.2.2.0 vjoy install only comes with x64.

Kuiriel, 2026-10-30:
I tried version vjoy 2.2.1-signed instead, which repeatedly installs fully but claims to have failed the install. That has an x86 folder. I copied the file across. It didn't load inside Opentrack. I reinstalled the new version of vjoy from here, 2.2.2.0 and the x86 folder in vjoy was deleted during the installation process. The vjoy option remains lost within Opentrack.

I had to uninstall vjoy, install vjoy 2.1.9.1 and copy the vjoyInterface.dll file from vjoy's x64 folder into Opentrack\Modules

Now I can see that my motions in Opentrack are being translated into vjoy axis movement (looking in usb game controllers on Windows)