opentrack/opentrack #1918: Support for Tobii Eye Tracker 5 in Linux

royborgen, 2024-08-19:
I managed to get Tobii Eye Tracker 5 working on Ubuntu using Talon: https://talon.wiki/.
However this alone does not make it work in games. It would be great if you guys could find a way to use Tobii Head Tracker 5 under Linux as a capture source and then translate this into TrackIR.

Priton-CE, 2024-09-04:
Could you explain how you did this? Tobii does not provide an official drivers.

royborgen, 2024-09-09:
Here is what I did to get it working, it was quite easy once I found the actual download and figured out that it did not support Wayland. You need to log in to an X11 session from your login manager.

I also installed the Talon tray icon. For stock Gnome, you need to install [AppIndicator and KStatusNotifierItem Support](https://extensions.gnome.org/extension/615/appindicator-support/) to see it. 

I downloaded Talon from https://talonvoice.com. After that I just ran the run.sh script. 

Official install instructions can be found here: https://talonvoice.com/docs/

Official github repo: 
https://github.com/talonhub/community

Hope that helps.

elkamynante, 2025-03-23:
Hello guys! I'm recently on Garuda Dragonized (after only 1 month on Mint). Neither in Mint or Garuda my Tobii 5 works and I can not find anything for help. Have you find something meanwhile?

sthalik, 2025-08-28:
See also #2067 and #2072.

zomgugoff, 2025-11-22:
@JustCryen As long as you have a Debian-based distro, that's still an option. The install script is for those distros. For Arch, there is only an AUR package for the manager. I've read about some Arch use in one of the issue threads, but it's not exactly clear how the installation is done for it.

Aetherall, 2026-04-05:
i have many values that dont change and stay 0... 
@megagtrwrath are those the same "missing features" you were refering to ?

[image]

megagtrwrath, 2026-04-05:
The missing features that were being discussed was related to the Head_Pose data stream, which it appears is calculated by the Tobii Experience runtime on Windows and isn't provided by the ET5 itself.

I would prefer to have a Linux supporting and open source Tobii/Eye tracker Runtime for full head and eye tracking but I don't have the skill set for that, especially not reverse engineering.
Would at least then be something that can be upstreamed to Opentrack too.