opentrack/opentrack #1922: Well darn : "Error occurred while loading tracker PointTracker 1.1  Failed to open camera ''

5YNTH3T1K, 2024-08-27:
So there I was thinking I would love to fly around Chernarus in my Ah-6j.... 

Loaded Arma2 , set my controls, tweaked my controller.  

OK lets see if I can get Opentrack to work.... 

and.... Nope. 

So I did the special dance with Libusbk , I can see a driver in Device Manager. 

Open track can't open the camera.... 

Yes I am using windows 7. 
Opentrack 2023.2  and 2023.3 and 2024.... 
I have tried LibusbK and LibusbO and the other one... 
etc etc. 

So yeah here I am again... unable to use the thing... 

Help? 

:- )

5YNTH3T1K, 2024-08-27:
Went back to CL Eye driver ( yes I actually paid for it years ago ) . 
Installed Opentrack 2021.1.3 

Worked as expected.  

both 2023.2 and 2023.3 refuse to work....

Old, but still works... like me...

5YNTH3T1K, 2024-08-28:
OK I have additional insight: 

IF I have the CL Eye Test program running, AND I try to run OpenTrack 2021.1.3, I get the same 
error message regarding the CAMERA not being present etc as I get when trying to run 2023.2 and up. 

Could this be the same/similiar issue that is stopping 2023.2 ( with Libusb driver ) and up from working on my machine? 
The camera cannot be found by OpenTrack 2023.2 etc, BECAUSE it is being used by something else, 
OR Opetrack thinks it is in use...  

This maybe a driver issue by LibUSB .... it's not actually working properly OR OpenTrack is not 
talking to it properly. 

The system is working fine with 2021.1.3 and the CL Eye driver.  

There is a vague possibility that LibUSB requies something that is not present on my machine and 
is not considered an error. 
Frankly I think that LibUSB is suspect and anything that has to make something up as it goes along 
could be introducing problems...  why not just make a damn driver that works and be done with it? 
**Is this some massive work around to avoid legal issues ?**  I just don't get it. 
Why all the hoopla ?  Can we driver already now?  Is it that much of deal ?  I am interested to find out. 

Well thanks for reading this! 

Back to flying... 

Cheers!

sthalik, 2024-08-28:
Try running the mode test executable from within a terminal and see if you get any error messages.

5YNTH3T1K, 2024-08-28:
You mean with 2023.2 and above?  
So... I will remove the CL eye driver and try the LibUSB drivers .... 
lemme just check that I know what a console window is etc. 
I'm going to try this all now.  So I will get back to you in a timely manner.

5YNTH3T1K, 2024-08-28:
ok so I ran ps3eye-mode-test.exe  for 2023.3 WITH the CL Eye driver. 

C:\>cd opentrack 2023.3

C:\opentrack 2023.3>cd modules

C:\opentrack 2023.3\modules>ps3eye-mode-test
no camera

Is this the kid of thing you mean ?  

I can now try with the LibUSB driver , after I uninstall Cl Eye and reinstall LibUSB ( I deleted them...) 

Am I doing ok ?

5YNTH3T1K, 2024-08-28:
oops the Camera was not plugged in... darn ...

5YNTH3T1K, 2024-08-28:
ok... so here we go with the CL Eye driver and the the ps3eye-mode-test.exe from 2023.3 

This is a test to see that I am doing this right . Some feed back would be great. 

C:\opentrack 2023.3\modules>ps3eye-mode-test
libusb: info [cache_config_descriptors] could not access configuration descripto
r 0 (dummy) for 'USB\VID_8087&PID_8008\5&A60DAA7&0&1': [31] A device attached to
 the system is not functioning.
[GOOD] 320x240@30Hz
[GOOD] 320x240@37Hz
[GOOD] 320x240@40Hz
[GOOD] 320x240@50Hz
[GOOD] 320x240@60Hz
[GOOD] 320x240@75Hz
[GOOD] 320x240@90Hz
[GOOD] 320x240@100Hz
[GOOD] 320x240@125Hz
[GOOD] 320x240@137Hz
[GOOD] 320x240@150Hz
[GOOD] 320x240@187Hz
[GOOD] 320x240@205Hz
[GOOD] 640x480@15Hz
[GOOD] 640x480@30Hz
[GOOD] 640x480@40Hz
[GOOD] 640x480@50Hz
[GOOD] 640x480@60Hz
[GOOD] 640x480@75Hz
[GOOD] 640x480@83Hz

C:\opentrack 2023.3\modules>

OK? I will now try the LibUSB drivers ( I rescued them from the recycle bin...)

sthalik, 2024-08-28:
No need. It's already possible to assign the driver to the device as shown by the successful run of the mode test.

You might want to try connecting the camera to the other USB controller. PCs normally have two, one has black connectors and the oner one has blue.

For the next version I'm going to improve error output for the open driver, but that's going to be in a long while. Camera manipulation is done through `ps3eye-subprocess.exe`.but it's difficult to run it manually.

5YNTH3T1K, 2024-08-28:
ok so I used the LibUSB driver and the 2023.3  ps3eye-mode-test.exe and I got this: ( in quote marks) 

"
[GOOD] 320x240@30Hz
[GOOD] 320x240@37Hz
[GOOD] 320x240@40Hz
[GOOD] 320x240@50Hz
[GOOD] 320x240@60Hz
[GOOD] 320x240@75Hz
[GOOD] 320x240@90Hz
[GOOD] 320x240@100Hz
[GOOD] 320x240@125Hz
[GOOD] 320x240@137Hz
[GOOD] 320x240@150Hz
[GOOD] 320x240@187Hz
[GOOD] 320x240@205Hz
[GOOD] 640x480@15Hz
[GOOD] 640x480@30Hz
[GOOD] 640x480@40Hz
[GOOD] 640x480@50Hz
[GOOD] 640x480@60Hz
[GOOD] 640x480@75Hz
[GOOD] 640x480@83Hz

C:\opentrack 2023.3\modules> "

BUT when I run Opentrack 2023.3 the camera fails to open. 

Mate.

5YNTH3T1K, 2024-08-28:
hmm. ok lemme see if changing USB ports does anything.

5YNTH3T1K, 2024-08-28:
Tried both Blue and Black USB ports. Same failure. 

**BUT .... 2021.1.3 works fine with the LibUSB driver.**  

So the Driver is working. 

Hope this helps! Keep up the good work, you are a STAR !!!! 

: - )