opentrack/opentrack #1923: RFC: Extend Proton support beyond steam to keep compatability with launchers (like Lutris) that use UMU

Priton-CE, 2024-09-04:
### The Problem
Lutris is starting to adapt [UMU](https://github.com/Open-Wine-Components/umu-launcher) which utilizes Proton for execution.

This is causing problems with titles run through Lutris with the UMU backend, as now instead of normal wine **they are using Proton for execution which Opentrack only supports in the context of Steam**.

(See #1869 as an example and for the current workaround which utilizes the current proton launch method.)

### The current state of Wine/Proton Support
Currently the proto-wine module supports two ways of launching the wrapper program inside of a wine/proton prefix:

1. Take a path to the wine executable and a path to the prefix and launch the wrapper script using these parameters. _(Optionally Opentrack also looks through the Lutris Runner Directory to suggest possible wine paths)_
2. Take a proton path and a steam_id to launch the wrapper using the proton path (and proton environment variables) inside the wine prefix corresponding to the steam_id. _(To find a proton path Opentrack looks through the `.../steamapps/common` folder as it houses the official Proton builds)_

What is to note here is that

- When Proton is used by other Game Launchers (like Lutris which it can now do thanks to UMU) Opentracks Wrapper Program will not be able to interface with them as its limited to run inside steams prefixes.
- Only official Proton Versions can be used as 3rd party Proton will not be installed to the `.../steam/steamapps/common` folder. This limits the use of 3rd party Proton versions like the ones made by Glorious Eggroll within Steam and also severely limiting possible support for UMU launched titles, as it prefers using GEs proton builds.

### What needs to be done
To support Proton games launched with Lutris one would need to allow the proton method to use proton versions and wine prefixes other than the ones created by steam (decoupling the proton feature from the steam echosystem but not destroying support).
That means:

1. The Dropdown to select Proton Versions would need to also contain the results from `.../steam/compatibilitytools.d/`
2. Instead of just taking a Steam Application ID the user would be able to select either the steam_id option or a custom path to a wine prefix

(This assumes that the current proton launch method can launch the wrapper in a way that allows it to communicate with UMU launched titles, which I dont see any reason why it should not that, especially after reports of a working workaround in the above issue)

### Conclusion
What are opinions on this topic?
Imo at least allowing the current proton method to search for proton versions utilized by UMU and allowing the use of custom prefixes beyond the steam_id method would be a good decision moving forward, as it improves the support for Opentrack with Steam titles that work better with 3rd Party Proton and it allows support for UMU launched titles.

Priton-CE, 2024-09-04:
I should note that I am writing this sort of on behalf of the Linux Users Group of Star Citizen and would at least try to implement these changes myself if approved.

muff11n, 2024-09-14:
not from SC community, but was trying to solve this issue too, so far no luck ( as wine-ge development will probably stop because of umu)

would be awesome if we had custom proton GE support
.../steam/compatibilitytools.d/

and WINE PREFIX instead of appID in this case, so we can connect opentrack to that proton instance

Priton-CE, 2024-09-20:
Small Update:
An experimental branch is available here that should be implementing the above changes https://github.com/Priton-CE/opentrack-StarCitizen/tree/wine-extended-proton (yes it says Star Citizen but it should be working for any UMU launched title).

This is the new proto/wine UI:
[image]

cezdro, 2024-09-21:
@Priton-CE In my case, it duplicated Protons, probably because of symlinks:

[image]

Priton-CE, 2024-09-21:
I think I pushed a change without testing it. For me its duplicating as well, so I will take care of that rq.

Tho for me the Steam Play Application ID is behaving completely fine.

Priton-CE, 2024-09-21:
@cezdro issues should be resolved. However I have no clue what could be causing that issue with the id selection as its working completely fine for me.
If there is anything interfering it must be some part of Opentracks settings framework.

closed 2024-10-03 by Priton-CE

Enovale, 2025-03-13:
Currently, non-steam games run through proton, through steam, cannot be added to this new UI because they have app ids that are beyond the size limit of the number wheel in the Qt GUI. @Priton-CE

Priton-CE, 2025-03-13:
So you use Steam as a launcher and that gives an ID that is too big?

Enovale, 2025-03-14:
> So you use Steam as a launcher and that gives an ID that is too big?

Yes. Non-steam games are `long` size numbers, e.g. `4252521017` or `3431493755`.

[This fork](https://github.com/JT8D-17/opentrack) seems to avoid the issue since it dynamically grabs the steam prefix data instead of being reliant on the Qt numeric spinner that ties the application to using integers. Unfortunately it also seems to be missing binaries so I can't build it atm.

sthalik, 2025-03-16:
What should be done is switching from a named memory mapping to a file-backed memory mapping. Make new NPClient{,64}.dll, stuff it in the right place in the prefix along with registry entries. Then opentrack doesn't need to integrate anything winelib-related and all the complexity can be simply dropped.