opentrack/opentrack #1933: New drive install

Cpt-Brewski, 2024-09-24:
I recently removed some drives from my PC and reinstalled Opentracker. Unfortunately it is still trying to detect the old file path for .ini profiles and I cant figure out how to change it. 

[image]
[image]

Len62, 2024-10-02:
Are you using the stand alone, or the installer version?

sthalik, 2024-10-02:
[image]

Cpt-Brewski, 2024-10-02:
> Are you using the stand alone, or the installer version?

Installer version

Cpt-Brewski, 2024-10-02:
> ![image](https://private-user-images.githubusercontent.com/1896811/372920388-46ac0238-5bfd-44fe-b920-9d9de7deeae6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc4OTEzMzksIm5iZiI6MTcyNzg5MTAzOSwicGF0aCI6Ii8xODk2ODExLzM3MjkyMDM4OC00NmFjMDIzOC01YmZkLTQ0ZmUtYjkyMC05ZDlkZTdkZWVhZTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDJUMTc0MzU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWE2M2Q5NWQ0MzY4Mzg2NzdmYTVjODg1YzQyYzVjNWRkMDAxMmM5YmQ2NTkwMTU0YTZmNjFhYjRkODVmODEyMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xWQMNsECI9DSVnSyb3XiefjbCn3-t5C45T-jGh_oAWg)

That doesnt work because the drive its looking for no longer exists

sthalik, 2024-10-02:
https://www.diskpart.com/windows-10/change-my-documents-location-windows-10-registry-1503.html

Cpt-Brewski, 2024-10-04:
> https://www.diskpart.com/windows-10/change-my-documents-location-windows-10-registry-1503.html

Spot on mate <3