opentrack/opentrack #1938: Vjoy not responding to Opentrack.

Keltskiy, 2024-10-02:
Currently using Eyeware beam for input, and Joystick emulation ~~ vjoystick as output.

I can get it to start without any error notices but looking at vJoy Monitor I'm not seeing a response from vjoy.

(Error on both Opentrack 2023.3.0 and 2024.1.1)
(Current vjoy installion is Jshaufers 2.1.9.1)

NathanielLLally, 2024-10-05:
dup of https://github.com/opentrack/opentrack/issues/1917

Keltskiy, 2024-10-05:
Not Dupe of #1917 

Open track starts process and proceeds as normal with 0 error notifications and displays, Raw tracker and Game Data are updating as normal inside opentrack.

vJoy however is unaffected.

Keltskiy, 2024-10-10:
update: was able to confirm opentrack is taking control of vjoy#1.
"vjoy feeder is unable to control vjoy#1 when opentrack is in use."

still not seeing any of the axis in vjoy#1 updating in response to input from opentrack though?

Keltskiy, 2024-10-11:
SOLUTION!!!
vjoyinterface.dll was mismatched to current/active vjoy install.

easiest solution:
copy vjoyinterface.dll from current/active vjoy install (*install location*/vJoy/x64/vjoyinterface.dll)
paste and replace existing file in (*install location*/opentrack/modules/vjoyinterfrace.dll)

closed 2024-10-11 by Keltskiy

negro9719, 2024-10-12:
I had the same issue with Vjoy 2.1.9.1(suposed to be working on win 11 #1700 

Copying vJoyInterface.dll from the vjoy folder to OpenTrack modules folder fixes the issue.

Opentrack Version used: 2024.1.1
vJoy version used: 2.1.9.1