opentrack/opentrack #1941: Wine output doesn't work with UMU Proton using opentrack-git on Manjaro

Gontran08, 2024-10-03:
Using the following settings:
[image]
[image]

I get this error:
[image]
[image]

I can confirm that the file exists:
[image]

I have the following version of opentrack:
[image]

It starts if I use "libevdev joystick receiver" as Output.

sthalik, 2024-10-03:
@Priton-CE?

Priton-CE, 2024-10-04:
Curious.

That would mean you have Steam installed at a location we are not expecting.

@Gontran08 could you verify that none of these paths a valid?

[image]

And if so could you kindly locate your `steam-runtime` folder?

Gontran08, 2024-10-04:
I'm using Flatpak for Steam and Lutris. I'm not sure why I have Proton in both the default path and the Flatpak path since I used ProtonUp-Qt to install GE-Proton and it cleary indicates the Flatpak directory. My guess is that Lutris installed them maybe ?

I didn't specify that I was using Flatpak, because I didn't think it could cause problem at this point. Why do we need a Steam Runtime for a non Steam game using UMU ?

Flatpak isn't supported currently ?

Priton-CE, 2024-10-04:
I will look into it more when I get home.

I wasn't the contributor that added Proton support so I am not sure why we need it but I will check if we can run without.

In the past we had trouble with Flatpack. #1879 should take care of that but for now, especially with Proton on the master branch, we have no Flatpack support.

sthalik, 2024-10-04:
Adding separate flatpak support seems like overkill.

With #1879 you'll be able to add the registry keys using a provided `.reg` file by running it in wine within flatpak, then write to the mapping through a memory-mapped file without having to invoke a `.dll.so` using Wine at all, just providing the Wine prefix's location.

Priton-CE, 2024-10-04:
I would also be against hardcoding any more Flatpack directories. Its already bad enough I did it for detecting Lutris runners. In the end Flatpack rn is too high of a chance for not working at all.

Sadly I dont have time tonight to investigate this properly so I will have to push this to tomorrow. Tho @Gontran08 the quick fix is either installing Steam via your native package manager or just copying the Steam Scout Runtime to one of the paths above.

Gontran08, 2024-10-04:
I just created a link in my home directory to use ubuntu12_32 in the Flatpak directory and it starts without throwing any error.

Gontran08, 2024-10-04:
Since a Flatpak doesn't have access to /usr/libexec, the path written in the registry doesn't work for my case though. It may not be the only issue with Flatpak anyway. For now, I'm too lazy to reconfigure everything to use a native version of Lutris. I'll just wait for the improvements you proposed.