opentrack/opentrack #1942: Install FSX/Prepar3D SimConnect SDK error

elimumm, 2024-10-04:
I am getting this message:

_Error occurred while loading protocol Microsoft FSX SimConnect

Install FSX/Prepar3D SimConnect SDK._

I downloaded Prepar3D_v6_SDK_Setup_6.1.10.31609 and MSFS_SDK_Core_Installer_0.24.3.0 . Still no luck. I am windows 11 and I believe I downloaded MSFS off the xbox/windows store on my PC.

0 replies

zhengwy888, 2024-11-12:
use the 'freetrack 2.0' option with MSFS.

pbix, 2024-11-19:
I have the same issue as reported by the OP.  There are no error messages reported by when I install the requested SimConnect SDK.  There are several versions available for download at https://www.prepar3d.com/sdk/#prepar3d-v6-sdk.  Which versions should I be using?

Attempting to start tracking in the Opentrack dialog results in the pop up error message:
"_Error occurred while loading protocol Microsoft FSX SimConnect

Install FSX/Prepar3D SimConnect SDK._"

I tried the "freetrack 2.0" option mentioned by zhengwy888 but I am unable to get the X,Y,Z translation function to work in opentrack and wonder if this is the reason.

I am using the Steam version of MSFS under windows 11.