opentrack/opentrack #1944: compiling error on ubuntu

MACWuppy, 2024-10-05:
I have been trying to get opentrack working on ubuntu 22.04 and have been only partially successful. I am stuck on trying to get the wine output available. I attempt to compile it as per the instructions [here](https://github.com/opentrack/opentrack/wiki/Building-on-Linux#building-on-debian-ubuntu-etc) with the added line of "ccmake ." and I have tried multiple times to compile and always get the same error `[ 65%] Generating opentrack-wrapper-wine.exe.so
/bin/sh: 1: wineg++: not found
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Error 127
make[1]: *** [CMakeFiles/Makefile2:2474: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make: *** [Makefile:136: all] Error 2`

It doesn't seem to matter what I do, I always get stuck at this point. Any help would be appreciated as I am at a loss at this point and its getting to be a little too technical for me.

Any suggestions?

Jer-C, 2024-10-07:
`wineg++: not found` seems to be a similar error as found in the final posts of https://github.com/opentrack/opentrack/issues/1662 - please make sure your wine installation is added to your PATH.

MACWuppy, 2024-10-16:
Is there a guide or a website I can go to that explains what exactly that means and how I do it? because I have followed several guides to install opentrack and I keep getting the same error.

MACWuppy, 2024-10-17:
OK, so after some much needed sleep and retrying, I have reinstalled wine and that error seems to be gone. I am now getting a different error now. the error is as follows: `/opentrack/compat/shm.h:10:10: fatal error: windows.h: No such file or directory
   10 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
winegcc: Segmentation fault (core dumped)`

I am unsure how to proceed with this as I have done some searching on the internet and have not found anything really that tells me anything about this error. am I missing something obvious?

sthalik, 2024-10-18:
If you compile Wine on your own then `winegcc` should work without problems.

MACWuppy, 2024-10-19:
Ok, not entirely sure what I did, but I got opentrack to compile and build. It works! the only thing now is I can't seem to get it to be seen by Elite Dangerous. Also when I try to set the appid under proton to Elite Dangerous' appid I get an error `Couldn't find a Wineprefix for AppId 359320` I am so close to getting this working (I hope), any other suggestions? Also thanks for all the help so far. I appreciate it.

MACWuppy, 2024-12-02:
Okay, so I have bit the bullet and reinstalled Elite onto the same drive as opentrack. I am not getting that error anymore, But I cannot get Opentrack to work with Elite still. It will automatically start when the game launches, but then it just sits there saying "waiting for game" Doesn't seem to matter what I try it doesn't want to play at all with Elite. I have been searching all over the internet in an attempt to sort this out and I have yet to find a solution that works. so with that, is there anywhere I can be pointed to? anything left to try? any help would be appreciated.

MACWuppy, 2024-12-10:
I'm not sure how but I finally got it working thanks to the script from [here](https://github.com/markx86/opentrack-launcher) Glad to finally have it working and be out in the black once more. closing this issue as its been solved.