opentrack/opentrack #1945: Head Center Offset Calibration Does Not Accept Pitch Movement

BluesAdam, 2024-10-06:
When trying to calibrate head center offset, instructions clearly state: "Use only yaw or pitch while calibrating. Don't roll or change position." However, it only accepts yaw and ROLL. I think this is causing faulty calibration. I didn't use the stable version so I don't know if this is specific to the June 2024 unstable version but wanted to create a bug report.

sthalik, 2024-10-06:
If you're only getting yaw and roll, your model is likely misconfigured or not correctly detected on screen.

BluesAdam, 2024-10-06:
Is there any way to reset or configure the model? All mapping is correct and works flawlessly in all games. Will double check but I don't know what else I can fix.

sthalik, 2024-10-06:
If it works on pitch in games then it's only the calibration that doesn't work. But calibration hasn't been updated in a long while.

When you see the red dot in mappings, the vertical axis is the raw input. On the pitch tab in the mappings dialog, look up how distinct the Y values are. The calibrator will only [save](https://github.com/opentrack/opentrack/blob/master/cv/translation-calibrator.hpp#L54-L55) an additonal sample when its inputs (yaw, pitch) vary by at least 2 and 1.5, respectively.

It's weird, you shouldn't be getting issues like that.

sthalik, 2024-10-06:
Made a typo in the previous post — raw input is the *horizontal* axis. I've edited the post but you might see the version with the typo in email notifications.

Sorry for the noises.

BluesAdam, 2024-10-06:
Mapping window also shows everything correctly. Both horizontal and vertical axes work correctly. Calibration simply will not accept pitch input but only accept roll input. I will try the stable version and let you know how it goes.

sthalik, 2024-10-06:
Since it's neuralnet then maybe @DaWelter has an idea.

DaWelter, 2024-10-11:
Hm .. I copy pasted most of the calibration code from the point tracker. Sounds like it's broken somehow.

I'll take a look soonish ... in the meantime, just work around it by "manual" calibration.

To this end, the visualization in the NN tracker should show a big red dot which is the head center. Set it's offsets so that it doesn't move (a lot) when you tilt/pitch your head.

closed 2024-10-13 by BluesAdam