opentrack/opentrack #1948: documentation for ubuntu missing info

p-s-t, 2024-10-13:
documentation for compiling on ubuntu
https://github.com/opentrack/opentrack/wiki/Building-on-linux
is missing line 

ccmake .

thus when compiling by begginers of most common linux distro
folk are clueless why are the missing the wine output by default

cgiAlexis, 2024-10-27:
When doing the `ccmake .` step I then got
``` bash
...
[ 70%] Generating opentrack-wrapper-wine.exe.so
/bin/sh: 1: wineg++: not found
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Error 127
make[1]: *** [CMakeFiles/Makefile2:2472: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

p-s-t, 2024-10-27:
Man i think i had same issue as you 
Hang on...
From my terminal history i used this command 
sudo apt install wine64-tools gcc-multilib g++-multilib

but from my browser history i visited this guide on building wine
https://gitlab.winehq.org/wine/wine/-/wikis/Building-Wine

But i do not see it in my terminal history. Nor do i remember doing it.
Try the first thing.