opentrack/opentrack #1964: Was working, now not

MysticDaedra, 2024-11-30:
Yesterday I got OpenTrack all set up with neuralnet for face tracking, calibrated etc. everything seemed to be working fine. Preview image, tracking was making the octopus move, etc. 

Today, after starting XP11 and starting OpenTrack, I am now getting a black screen in the preview window, and obviously no tracking is occurring. Webcam is a Logitech C920, and is working fine in all other programs. The LEDs come on as well, so I know that the camera is on and sending data to Opentrack.

Any thoughts? I'm using the 64-bit version for the extra NN performance, and... it _was_ working just fine, no idea why it stopped working. Any help would be appreciated.

DaWelter, 2024-12-01:
Is it only the NN tracker or also the pointtracker?

Does it help when you unplug the camera for a moment?

MysticDaedra, 2024-12-01:
Hi, sorry for the late reply. I uninstalled and reinstalled the drivers,
and it started working again, haven't had any similar issues since. Unsure
what was the cause, again considering the camera was working in other
programs. Regardless, it is working now.

As a note, it was as if the input from the camera was black/blank, so no
form of tracking would work. I'm not sure it was an NN issue to begin with,
to be honest.