opentrack/opentrack #1965: Failing To Connect To Tobii Eye Tracker 5

RichardPM7, 2024-12-01:
I am getting this message when I try to connect to my Tobii Eyetracker 5 in opentrack-2023.3.0.

'Failed to connect to tobii-ttp://IS5FF-100202739121'

Can you help please

RichardPM7, 2024-12-01:
Sorry I should have said that this occurs when I hit start.

mmosoll, 2025-01-01:
Only as info: Tobii Eyetracker 5 working fine with the OpenTrack pre-release version 2024.1.1

My config: Windows 11, Tobii Eyetracker 5 + Tobii Game Hub

moxiejeffdallas, 2025-01-17:
Same issue; came out of nowhere and now I get it every time.

mmosoll, 2025-01-17:
Hopefully not happening on my side, but the message seems more coming from a failling Tobii connection than OpenTrack.

Did you try to reinstall the USB driver from Tobii Original website:
https://gaming.tobii.com/getstarted/

Direct Download:
https://files.update.tech.tobii.com/Tobii.EyeTracker5.Offline.Installer_4.183.0.30025.msi

And try to change also the USB port used to connect your Tobii Eye Tracker 5

izipime, 2025-04-26:
I had the same issue but I just solved it this way

Open Tobii Gamehub
And, click the opentrack in the list
Then, click the small triangle next to its name and specify the executable file