opentrack/opentrack #1970: Doesn't compile from source on linux

DoomSlinger, 2024-12-23:
Get this error when trying to commpile from source on linux

~/opentrack/compat/process-list.hpp:153:29: error: ‘PIDS_VAL’ was not declared in this scope
  153 |         char  **p_cmdline = PIDS_VAL(rel_cmdline, strv,  stack, info);
      |                             ^~~~~~~~
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/build.make:210: gui/CMakeFiles/opentrack-user-interface.dir/process_detector.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2405: gui/CMakeFiles/opentrack-user-interface.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

sthalik, 2024-12-24:
There's something wrong with your `libproc2`. This is a public API function, see: https://gitlab.com/procps-ng/procps/-/blob/master/library/include/pids.h#L247

```c
#define PIDS_VAL( relative_enum, type, stack ) stack -> head [ relative_enum ] . result . type
```

Just add it as line 152.

By commenting out the feature check as below you can go back to libprocps if you're using an old distro that still has it.

```patch
diff --git gui/CMakeLists.txt gui/CMakeLists.txt
index 8c57221a..5db77c24 100644
--- gui/CMakeLists.txt
+++ gui/CMakeLists.txt
@@ -13,7 +13,7 @@ if(APPLE)
 elseif(LINUX)
     otr_pkgconfig_(has-libproc2 ${self} libproc2)
     if(has-libproc2)
-        target_compile_definitions(${self} PRIVATE -DOTR_HAS_LIBPROC2)
+        #target_compile_definitions(${self} PRIVATE -DOTR_HAS_LIBPROC2)
     else()
         otr_pkgconfig(${self} libprocps)
     endif()
```

krims0n32, 2024-12-24:
Same issue here, happened after procps-ng upgraded (arch):

DoomSlinger, 2024-12-24:
> Same issue here, happened after procps-ng upgraded (arch):
> 
> `procps-ng (4.0.4-3 -> 4.0.5-1)`

This was my problem too, I downgraded my packages to about a week ago and it compiled.

McTom234, 2024-12-24:
I can confirm: downgrading makes the build successful

thorger, 2024-12-25:
@sthalik, I have same issue, and as the API has changed as per 
https://gitlab.com/procps-ng/procps/-/blob/master/library/include/pids.h#L247, 
the code here is still calling the old signature:
`char  **p_cmdline = PIDS_VAL(rel_cmdline, strv,  stack, info);`
https://github.com/opentrack/opentrack/blob/master/compat/process-list.hpp#L153C23-L153C23
I get same error as the fn call is trying to pass info param which is not handled by the new libproc2, because the info param has been removed 7 months ago:
https://gitlab.com/procps-ng/procps/-/commit/967fdcfb06e20aad0f31073474cb94545c9bdea5
Is this something to be adjusted in the process-list.hpp?

sthalik, 2024-12-26:
Should be fixed in the newest revision. Ping me otherwise.

closed 2024-12-26 by sthalik

JT8D-17, 2025-06-19:
Patching process_list.cpp with that code above produces "error: ‘OPENTRACK_PIDS_VAL’ was not declared in this scope" on compilation.

cyber199, 2025-07-10:
Unsure if this is related (I'm an idiot) but as the issue is still open I'll show what's happening to me
```
$ make
[...]
[ 29%] Building CXX object csv/CMakeFiles/opentrack-csv.dir/csv.cpp.o
/media/path/to/opentrack/csv/csv.cpp:24:15: error: ‘QLatin1StringView’ was not declared in this scope; did you mean ‘QLatin1String’?
   24 | auto do_scanf(QLatin1StringView s, unsigned(&tmp)[8])
      |               ^~~~~~~~~~~~~~~~~
      |               QLatin1String
/media/path/to/opentrack/csv/csv.cpp:24:46: error: ‘tmp’ was not declared in this scope; did you mean ‘tm’?
   24 | auto do_scanf(QLatin1StringView s, unsigned(&tmp)[8])
      |                                              ^~~
      |                                              tm
/media/path/to/opentrack/csv/csv.cpp:24:53: error: expression list treated as compound expression in initializer [-fpermissive]
   24 | auto do_scanf(QLatin1StringView s, unsigned(&tmp)[8])
      |                                                     ^
/media/path/to/opentrack/csv/csv.cpp:25:1: error: expected ‘,’ or ‘;’ before ‘{’ token
   25 | {
      | ^
/media/path/to/opentrack/csv/csv.cpp: In function ‘bool getGameData(int, unsigned char*, QString&)’:
/media/path/to/opentrack/csv/csv.cpp:95:61: error: ‘QLatin1StringView’ was not declared in this scope; did you mean ‘QLatin1String’?
   95 |                 else if (id_cstr.length() != 22 || do_scanf(QLatin1StringView(id_cstr), tmp) != 11)
      |                                                             ^~~~~~~~~~~~~~~~~
      |                                                             QLatin1String
make[2]: *** [csv/CMakeFiles/opentrack-csv.dir/build.make:80: csv/CMakeFiles/opentrack-csv.dir/csv.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1531: csv/CMakeFiles/opentrack-csv.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
Partially obfuscated, not that you knowing my username and drive name is a big issue, but eh.
I'll see if I can figure out and try the above fixes, it shouldn't be too hard to find the files,but that's where I am.
It's also worth mentioning that I'm on Mint, which seems to do some funny stuff with it's packages.  I didn't have anything to do with `libproc2` in my default repos, and I had to install `qt6-l10n-tools` and `qt6-tools-dev-tools` as additional dependencies.  Maybe it serves me right for straying off the beaten track that is Ubuntu, haha, but you'd think Noob-Friendly Mint wouldn't be too different.

cyber199, 2025-07-10:
Poked around in the process-list file, unpatched, repatched, and removed the offending line, so in my unprofessional opinion it's not related.  Not sure what the griff is with the csv.cpp file, I'm barely literate in C++.  Would it be helpful to create a new issue, or keep my problem here?

sthalik, 2025-07-13:
What's your Qt version @cyber199?

commit 1a40b793 (2024-12-26): compat: fix Linux process list

`procps-ng` silently broke API compatibility in a patch release.

Reported-by: @DoomSlinger
Fix-by: @thorger
Fixes: #1970